ArduPilot/pymavlink #432: Typescript Mavlink Generator Has Non-Existant Import

FullPint, 2020-07-08:
In the `generator/mavgen_typescript.py` on **line 52, 65, 66**:
```python
registry_f.write("import {MAVLinkMessage} from 'node-mavlink';\n") # line 52 

f.write("import {MAVLinkMessage} from 'node-mavlink';\n") #65

f.write("import {readInt64LE, readUInt64LE} from 'node-mavlink';\n") #66
```
The import does not exist on npm. If this is reference to [omcaree/node-mavlink](https://github.com/omcaree/node-mavlink), there may need to be some sort of instruction on it being a dependency, and how to add it since it is not available on npm, and because the official docs reference the following package [ifrunistuttgart/node-mavlink]( https://github.com/ifrunistuttgart/node-mavlink). 

If this is actually in reference to the second package, it can be added if for each of the above from {package} was replaced with `'@ifrunistuttgart/node-mavlink';`

Unless there is something I'm missing -- I believe this is necessary for this to work.

rummanwaqar, 2020-09-01:
Hey do you know why line #66 is needed? Its not being used in any of these files

FullPint, 2020-09-02:
@rummanwaqar I believe it’s an import used in the sister package.

padcom, 2023-10-16:
> +1
> 
> The NPM package that is available under the name node-mavlink is ArduPilot's: https://github.com/ArduPilot/node-mavlink But that one does not contain the imports MAVLinkMessage, readInt64LE or readUInt64LE.

Hi guys, I am the maintainer of node-mavlink. What seems to be the issue with readInt64LE and readUInt64LE? Are those breaking some compatibility with other packages? If so you can always make a PR, I'll gladly review it!

rremigius, 2023-10-16:
Hi, thanks for your response! The issue is that MAVLinkMessage, readInt64LE and readUInt64LE do not seem to exist in node-mavlink, though the generated files try to import those:

import {MAVLinkMessage} from 'node-mavlink';
import {readInt64LE, readUInt64LE} from 'node-mavlink';

Is there a different version of node-mavlink that I am not aware of that does include this class and these functions? Or is node-mavlink referring to a different package than ArduPilot/node-mavlink, as the original issue poster suggested?

padcom, 2023-10-16:
The `node-mavlink` package that I maintain does not export any of those methods for low-level byte-by-byte reading of values. That's implemented in https://github.com/ArduPilot/node-mavlink/blob/master/lib/serialization.ts, it's an internal part of the system and will never be exposed. This brings me to the conclusion that it must be about some other package.

rremigius, 2023-10-17:
Yeah issue is about the pymavlink repository, which generates code that depends on some package called node-mavlink, but apparently it is not the one that npm installs when you `npm install node-mavlink` (the one you maintain), but also not clear which one it does depend on then.