ArduPilot/pymavlink #440: "error: command 'x86_64-linux-gnu-gcc' failed with exit status 1" when installing using PIP

rekhyzakaria, 2020-07-30:
### Error when installing pymavlink on kali linux using pip

Hi, please help me. I have installed pymavlink before and found no errors, but when I want to update the latest version of pymavlink with reinstalling it using pip, there comes the error. 

so, i'm using this `pip install pymavlink` (should be latest version)
and i got this

> Collecting pymavlink
  Using cached pymavlink-2.4.9.tar.gz (3.6 MB)
Requirement already satisfied: future in /usr/lib/python2.7/dist-packages (from pymavlink) (0.18.2)
Requirement already satisfied: lxml in /usr/lib/python2.7/dist-packages (from pymavlink) (4.5.0)
Building wheels for collected packages: pymavlink
  Building wheel for pymavlink (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-nFuepy/pymavlink/setup.py'"'"'; __file__='"'"'/tmp/pip-install-nFuepy/pymavlink/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-FzChkZ
       cwd: /tmp/pip-install-nFuepy/pymavlink/
  Complete output (187 lines):
  running bdist_wheel
  running build
  running build_py
  Using message definitions from /tmp/pip-install-nFuepy/pymavlink/message_definitions
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/icarous.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ualberta.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ASLUAV.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/test.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ardupilotmega.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/python_array_test.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/paparazzi.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/minimal.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/common.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/matrixpilot.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/uAvionix.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/slugs.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/standard.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/autoquad.xml for protocol 1.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/icarous.xml for protocol 2.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ualberta.xml for protocol 2.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ASLUAV.xml for protocol 2.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/test.xml for protocol 2.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/ardupilotmega.xml for protocol 2.0
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/python_array_test.xml for protocol 2.0                                                          
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/paparazzi.xml for protocol 2.0                                                                  
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/minimal.xml for protocol 2.0                                                                    
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/common.xml for protocol 2.0                                                                     
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/matrixpilot.xml for protocol 2.0                                                                
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/uAvionix.xml for protocol 2.0                                                                   
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/slugs.xml for protocol 2.0                                                                      
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/standard.xml for protocol 2.0                                                                   
  Building /tmp/pip-install-nFuepy/pymavlink/message_definitions/v1.0/autoquad.xml for protocol 2.0                                                                   
  creating build                                                                                                                                                      
  creating build/lib.linux-x86_64-2.7                                                                                                                                 
  creating build/lib.linux-x86_64-2.7/pymavlink                                                                                                                       
  copying ./quaternion.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                     
  copying ./__init__.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                       
  copying ./mavexpression.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                  
  copying ./mavparm.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                        
  copying ./mavwp.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                          
  copying ./mavutil.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                        
  copying ./DFReader.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                       
  copying ./mavextra.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                       
  copying ./fgFDM.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                          
  copying ./rotmat.py -> build/lib.linux-x86_64-2.7/pymavlink                                                                                                         
  creating build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                                             
  copying ./generator/mavgen_wlua.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                
  copying ./generator/mavgen_lua.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                 
  copying ./generator/mavgen_objc.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                
  copying ./generator/mavgen_javascript.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                          
  copying ./generator/mavtestgen.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                 
  copying ./generator/__init__.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                                   
  copying ./generator/mavgen_cpp11.py -> build/lib.linux-x86_64-2.7/pymavlink/generator                                                                               
[... 324 lines cut ...]
[... cut ...]
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:
The following packages have unmet dependencies:
 libc6-dev : Breaks: libgcc-9-dev (< 9.3.0-5~) but 9.2.1-19 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

what should i do? Please help me

peterbarker, 2020-07-31:
On Thu, 30 Jul 2020, rekhyzakaria wrote:

>   Error when installing pymavlink on kali linux using pip

You need to install libpython.

Were you following any docs that might need updating with a ".. note:"?

rekhyzakaria, 2020-07-31:
> On Thu, 30 Jul 2020, rekhyzakaria wrote: Error when installing pymavlink on kali linux using pip
> You need to install libpython. Were you following any docs that might need updating with a ".. note:"?

thanks for the reply. i really new at this. i just found the problem is in my kali linux, so i'm using this `sudo apt-get dist-upgrade` and then `dpkg --configure -a` to fix it. After that, i can install pymavlink without error. I will close this issue, Thankyou

closed 2020-07-31 by rekhyzakaria