ArduPilot/pymavlink #445: Contribution guide

ar13pit, 2020-08-17:
Hi,

Are there any guidelines for making a contribution to this repository?

I opened a PR about 2 weeks ago #441 but didn't know who to ask for a review. 

It will be great if some information about this can be added in the README.

Thanks

peterbarker, 2020-08-27:
On Mon, 17 Aug 2020, Arpit Aggarwal wrote:

> Are there any guidelines for making a contribution to this repository?

"Create a PR and wait way too long" pretty much covers it, sorry.

ar13pit, 2020-08-27:
> On Mon, 17 Aug 2020, Arpit Aggarwal wrote: Are there any guidelines for making a contribution to this repository?
> "Create a PR and wait way too long" pretty much covers it, sorry.

Haha, I understand it can take way too long for someone to have a look if no one is actually assigned to an issue or PR. As GitHub doesn't allow non-contributors to assign people to issues or PRs, it is possible for you guys to mention on the README who to tag in a comment to have issues or PRs atleast assigned to the right so that PRs are processed faster?

davidbuzz, 2020-12-11:
as a 100% volunteer codebaase, the other way to get your PR attention is to do a good job reviewing other people's PRs. :-)