ArduPilot/pymavlink #452: MAVLink Python decoder removes raw byte strings

len0rd, 2020-08-31:
Before considering a message fully decoded, python's `MAVLink.decode` goes and trims any extra 0x00's off the string to make sure its of the correct length. ([relevant code here](https://github.com/ArduPilot/pymavlink/blob/00177bd9f90e9b89febd94c6dec750eaa3c1186e/generator/mavgen_python.py#L871)). Before trimming, `MAVLink.to_string` is called in an attempt to put the raw bytes into some type of real ascii/utf-8 string value

This becomes an issue for messages that use char arrays as raw buffers (`PARAM_EXT_SET` for instance). If the `to_string` method fails to decode the raw bytes into a valid ascii/utf-8 string it tosses out the bad bytes and returns `_XXX` ([here](https://github.com/ArduPilot/pymavlink/blob/00177bd9f90e9b89febd94c6dec750eaa3c1186e/generator/mavgen_python.py#L750)), tossing out what is likely valid data given the CRC check has already passed.

Currently, my temporary solution is to not attempt any string conversion and only rstrip the extra 0's off the array and leave it to the user to change the value from bytes to string as needed. Something like this on L870:
`tlist[i] = tlist[i].rstrip(b'\x00')`

peterbarker, 2020-09-01:
On Mon, 31 Aug 2020, len0rd wrote:

> Before considering a message fully decoded, python's MAVLink.decode goes and trims any extra 0x00's off the string to make sure its of the correct length.
> (relevant code here). Before trimming, [...]

Why are those messages using char[] rather than uint8_t[]?  IIUC that 
value ought to be opaque until something-or-other looks at the type field?

len0rd, 2020-09-01:
Good point. I'm not sure why char[] is being used over uint8_t[]. I could open an issue on mavlink as well.

I think my main point is, when `to_string` is called, we're confident the bytes are valid since the crc has passed. From my (obviously biased) perspective it seems like if str conversion fails it should still preserve the received bytes for the user to work with.

len0rd, 2020-09-01:
Opening an issue on mavlink for this

closed 2020-09-01 by len0rd