ArduPilot/pymavlink #464: Serial connection fails sending mission to APM

pariaspe, 2020-09-23:
Hi, I'm trying to set a mission on my APM (arduplane) using pymavlink via a serial connection.

This is what I tried:

```python
from pymavlink import mavutil
master = mavutil.mavlink_connection(port, baudrate)
master.wait_heartbeat()
master.mav.request_data_stream_send(master.target_system,
                                        master.target_component,
                                        mavutil.mavlink.MAV_DATA_STREAM_ALL,
                                        RATE, 1)
```

I assume that this is working because I start receiving msgs from the APM. However, when I try to send a mission to the APM, I got stuck. 

Following the MAVLink [mission upload protocol](https://mavlink.io/en/services/mission.html#uploading_mission), after sending a `MISSION COUNT` I receive a `MISSION_REQUEST {target_system : 255, target_component : 0, seq : 0}` to which I am answering with the first `MISSION ITEM`. Instead of receiving the second request, I keep receiving the same `MISSION REQUEST` until I get a `MISSION_ACK {target_system : 255, target_component : 0, type : 1}` which stops the uploading.

What is going wrong? I got it working with PX4 but it fails with APM.

Thanks in advance, Pedro.

amilcarlucas, 2020-09-29:
The first item is the home location.

pariaspe, 2020-09-30:
Hi @amilcarlucas, thanks for your answer. 

I tried with your suggestion but it keeps failing.

davidbuzz, 2020-12-11:
if the vehicle is repeating the MISSION ITEM request, then its because according to teh spec, it didn't recieve a properly formatted  MISSION_ITEM  or  MISSION_ITEM_INT , i expect you've got something set wrong in the packet u are sending.  i recommend looking at an existing implementation in , say mavproxy or mission planner, and making sure your packet/s are formatted the same as one of the existing/working implementations.  Closing this now, as it appears to be a misconfiguration issue, not a bug.

closed 2020-12-11 by davidbuzz