ArduPilot/pymavlink #465: Request to add command to clear dataflash logs

LeonFargher, 2020-09-23:
Hi all,

Would it be possible to add a command compatible with dronekit to clear dataflash logs from the pixhawk (ardupilot/arduplane) rather than having to clear them through mission planner?

Commerical platforms that wouldn't have their logs cleared, would be useful to do it through onboard python script

andyp1per, 2020-09-23:
This is already supported by mavproxy and mavlink https://ardupilot.org/mavproxy/docs/modules/log.html

LeonFargher, 2020-09-23:
hi andy, 

dont understand how this works, where do you issue this command?

what device is receiving this. The pixhawk? and over what medium?

Thank you for your reply :)

LeonFargher, 2020-09-25:
Found the code to do it below

msg = vehicle.message_factory.log_erase_encode(0,0)
vehicle.send_mavlink(msg)

amilcarlucas, 2020-09-29:
Solved

closed 2020-09-29 by amilcarlucas