ArduPilot/pymavlink #477: Python: generated code is broken with Python 3.9

ntamas, 2020-11-12:
It seems like the generated modules in `pymavlink.dialects` is broken on Python 3.9.

*Steps to reproduce:*

1. Create a clean Python 3.9 virtualenv.
2. Install `pymavlink` from PyPI.
3. Run the following code snippet:

```
from pymavlink import mavutil
from pymavlink.dialects.v10.ardupilotmega import MAV_TYPE_GENERIC, MAV_AUTOPILOT_GENERIC

conn = mavutil.mavlink_connection('udpin:localhost:14540')
conn.mav.heartbeat_encode(MAV_TYPE_GENERIC, MAV_AUTOPILOT_GENERIC, 0, 0, 0).pack(conn.mav)
```

*Expected outcome:*

The encoded byte-level representation of a MAVLink heartbeat packet (which works on Python 3.8)

*Actual outcome:*

```
Traceback (most recent call last):
  File ".venv/lib/python3.9/site-packages/pymavlink/generator/mavcrc.py", line 35, in accumulate_str
    bytes.fromstring(buf)
AttributeError: 'array.array' object has no attribute 'fromstring'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test.py", line 5, in <module>
    conn.mav.heartbeat_encode(MAV_TYPE_GENERIC, MAV_AUTOPILOT_GENERIC, 0, 0, 0).pack(conn.mav)
  File ".venv/lib/python3.9/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 7375, in pack
    return MAVLink_message.pack(self, mav, 50, struct.pack('<IBBBBB', self.custom_mode, self.type, self.autopilot, self.base_mode, self.system_status, self.mavlink_version), force_mavlink1=force_mavlink1)
  File ".venv/lib/python3.9/site-packages/pymavlink/dialects/v10/ardupilotmega.py", line 243, in pack
    crc.accumulate_str(struct.pack('B', crc_extra))
  File ".venv/lib/python3.9/site-packages/pymavlink/generator/mavcrc.py", line 37, in accumulate_str
    bytes.frombytes(buf.encode())
AttributeError: 'bytes' object has no attribute 'encode'
```

The bug can be resolved by replacing `crc.accumulate_str` with `crc.accumulate` in the generated dialect file, but this may break older Python version (not sure, I haven't tested it).

ntamas, 2020-11-12:
It looks like this issue is addressed by https://github.com/ArduPilot/pymavlink/pull/449/ -- looking forward to seeing that PR get merged soon.

peterbarker, 2020-11-13:
Tagged PR was merged - thanks for stirring this up again :-)

closed 2020-11-13 by peterbarker