ArduPilot/pymavlink #486: pip install fails for python >= v3.7 in docker

jonaspetersorensen, 2020-12-06:
The documentation says
> It is recommended to install pymavlink from PyPI with pip, that way dependencies should be auto installed by pip.
> sudo python -m pip install --upgrade pymavlink

### Using python v3.6 OK
Create a dockerfile with the following content,
```Dockerfile
FROM python:3.6

RUN python -m venv /opt/venv && . /opt/venv/bin/activate
ENV PATH="/opt/venv/bin:$PATH"
WORKDIR /opt/venv

RUN pip install --upgrade pymavlink
```
Then build it,
```sh
docker build -t test .
```
#### Result
It will build fine.

### Using python >= v3.7 FAILS
Update and build the dockerfile from previous example,
```Dockerfile
FROM python:3.7
...
```

#### Result
The pip install of pymavlink fails,
```sh
Downloading pymavlink-2.4.14.tar.gz (4.1 MB)
    ERROR: Command errored out with exit status 1:
     command: /opt/venv/bin/python -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-uxdat5ec/pymavlink/setup.py'"'"'; __file__='"'"'/tmp/pip-install-uxdat5ec/pymavlink/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-4ausdmc0
         cwd: /tmp/pip-install-uxdat5ec/pymavlink/
    Complete output (44 lines):
    WARNING: The wheel package is not available.
      ERROR: Command errored out with exit status 1:
       command: /opt/venv/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-wheel-6f9ect9_/future/setup.py'"'"'; __file__='"'"'/tmp/pip-wheel-6f9ect9_/future/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-aplvpm4y
           cwd: /tmp/pip-wheel-6f9ect9_/future/
      Complete output (6 lines):
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help

      error: invalid command 'bdist_wheel'
      ----------------------------------------
      ERROR: Failed building wheel for future
    ERROR: Failed to build one or more wheels
    Traceback (most recent call last):
      File "/opt/venv/lib/python3.7/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/usr/local/lib/python3.7/subprocess.py", line 363, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/opt/venv/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp7f5svmnk', '--quiet', 'future']' returned non-zero exit status 1.

    During handling of the above exception, another exception occurred:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-uxdat5ec/pymavlink/setup.py", line 181, in <module>
        ext_modules = extensions
      File "/opt/venv/lib/python3.7/site-packages/setuptools/__init__.py", line 143, in setup
        _install_setup_requires(attrs)
      File "/opt/venv/lib/python3.7/site-packages/setuptools/__init__.py", line 138, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/opt/venv/lib/python3.7/site-packages/setuptools/dist.py", line 698, in fetch_build_eggs
        replace_conflicting=True,
      File "/opt/venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 783, in resolve
        replace_conflicting=replace_conflicting
      File "/opt/venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1066, in best_match
        return self.obtain(req, installer)
      File "/opt/venv/lib/python3.7/site-packages/pkg_resources/__init__.py", line 1078, in obtain
        return installer(requirement)
      File "/opt/venv/lib/python3.7/site-packages/setuptools/dist.py", line 754, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/opt/venv/lib/python3.7/site-packages/setuptools/installer.py", line 130, in fetch_build_egg
        raise DistutilsError(str(e))
    distutils.errors.DistutilsError: Command '['/opt/venv/bin/python', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/tmp/tmp7f5svmnk', '--quiet', 'future']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

mrpollo, 2020-12-16:
I can confirm the same issue on macOS Big Sur (11.0.1) using Python 3.9

```
$ pip3 install --user pymavlink

Collecting pymavlink
  Using cached pymavlink-2.4.14.tar.gz (4.1 MB)
    ERROR: Command errored out with exit status 1:
     command: /Users/rroche/.pyenv/versions/3.9.0/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-install-7m8io9uy/pymavlink_b7195e3570134cc4889f76713cc7c6e2/setup.py'"'"'; __file__='"'"'/private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-install-7m8io9uy/pymavlink_b7195e3570134cc4889f76713cc7c6e2/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-pip-egg-info-o8diu6d9
         cwd: /private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-install-7m8io9uy/pymavlink_b7195e3570134cc4889f76713cc7c6e2/
    Complete output (44 lines):
    WARNING: The wheel package is not available.
      ERROR: Command errored out with exit status 1:
       command: /Users/rroche/.pyenv/versions/3.9.0/bin/python3.9 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-wheel-klhl_g9t/future_b5f0ee4a497442a2ad5c377f6b6b547a/setup.py'"'"'; __file__='"'"'/private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-wheel-klhl_g9t/future_b5f0ee4a497442a2ad5c377f6b6b547a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-wheel-vymdi6tx
           cwd: /private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-wheel-klhl_g9t/future_b5f0ee4a497442a2ad5c377f6b6b547a/
      Complete output (6 lines):
      usage: setup.py [global_opts] cmd1 [cmd1_opts] [cmd2 [cmd2_opts] ...]
         or: setup.py --help [cmd1 cmd2 ...]
         or: setup.py --help-commands
         or: setup.py cmd --help

      error: invalid command 'bdist_wheel'
      ----------------------------------------
      ERROR: Failed building wheel for future
    ERROR: Failed to build one or more wheels
    Traceback (most recent call last):
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/installer.py", line 126, in fetch_build_egg
        subprocess.check_call(cmd)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/subprocess.py", line 373, in check_call
        raise CalledProcessError(retcode, cmd)
    subprocess.CalledProcessError: Command '['/Users/rroche/.pyenv/versions/3.9.0/bin/python3.9', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/tmp_qg7_drw', '--quiet', 'future']' returned non-zero exit status 1.

    The above exception was the direct cause of the following exception:

    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/pip-install-7m8io9uy/pymavlink_b7195e3570134cc4889f76713cc7c6e2/setup.py", line 109, in <module>
        setup (name = 'pymavlink',
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/__init__.py", line 152, in setup
        _install_setup_requires(attrs)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/__init__.py", line 147, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/dist.py", line 673, in fetch_build_eggs
        resolved_dists = pkg_resources.working_set.resolve(
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/pkg_resources/__init__.py", line 764, in resolve
        dist = best[req.key] = env.best_match(
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/pkg_resources/__init__.py", line 1049, in best_match
        return self.obtain(req, installer)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/pkg_resources/__init__.py", line 1061, in obtain
        return installer(requirement)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/dist.py", line 732, in fetch_build_egg
        return fetch_build_egg(self, req)
      File "/Users/rroche/.pyenv/versions/3.9.0/lib/python3.9/site-packages/setuptools/installer.py", line 128, in fetch_build_egg
        raise DistutilsError(str(e)) from e
    distutils.errors.DistutilsError: Command '['/Users/rroche/.pyenv/versions/3.9.0/bin/python3.9', '-m', 'pip', '--disable-pip-version-check', 'wheel', '--no-deps', '-w', '/var/folders/9l/sz9qr9s52_36g5y5ngd73j6r0000gn/T/tmp_qg7_drw', '--quiet', 'future']' returned non-zero exit status 1.
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

mrpollo, 2020-12-16:
I resolved this issue by installing the `wheel` package

```
pip3 install --user wheel
pip3 install --user pymavlink
```

AndKe, 2021-03-18:
There is something more to it on 3.9.2  - wheel alone does not fix it,  lxml fails.
```
pi@raspberrypi:~ $ pip install  pymavlink
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pymavlink
  Using cached pymavlink-2.4.14.tar.gz (4.1 MB)
Collecting future
  Using cached https://www.piwheels.org/simple/future/future-0.18.2-py3-none-any.whl (491 kB)
Collecting lxml
  Using cached lxml-4.6.2.tar.gz (3.2 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.pyenv/versions/3.9.2/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-e6s2v0e0/lxml_510e822d85e3424a93786e988c96b77a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-e6s2v0e0/lxml_510e822d85e3424a93786e988c96b77a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-n63d5h91
         cwd: /tmp/pip-install-e6s2v0e0/lxml_510e822d85e3424a93786e988c96b77a/
    Complete output (3 lines):
    Building lxml version 4.6.2.
    Building without Cython.
    Error: Please make sure the libxml2 and libxslt development packages are installed.
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/db/f7/43fecb94d66959c1e23aa53d6161231dca0e93ec500224cf31b3c4073e37/lxml-4.6.2.tar.gz#sha256=cd11c7e8d21af997ee8079037fff88f16fda188a9776eb4b81c7e4c9c0a7d7fc (from https://pypi.org/simple/lxml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, != 3.4.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached lxml-4.6.1.tar.gz (3.2 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.pyenv/versions/3.9.2/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-e6s2v0e0/lxml_d88deb299e7f42e38d4e5085b363f0fd/setup.py'"'"'; __file__='"'"'/tmp/pip-install-e6s2v0e0/lxml_d88deb299e7f42e38d4e5085b363f0fd/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-vaafm7by
         cwd: /tmp/pip-install-e6s2v0e0/lxml_d88deb299e7f42e38d4e5085b363f0fd/
    Complete output (3 lines):
    Building lxml version 4.6.1.
    Building without Cython.
    Error: Please make sure the libxml2 and libxslt development packages are installed.
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/c5/2f/a0d8aa3eee6d53d5723d89e1fc32eee11e76801b424e30b55c7aa6302b01/lxml-4.6.1.tar.gz#sha256=c152b2e93b639d1f36ec5a8ca24cde4a8eefb2b6b83668fcd8e83a67badcb367 (from https://pypi.org/simple/lxml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, != 3.4.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached lxml-4.6.0.tar.gz (3.2 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.pyenv/versions/3.9.2/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-e6s2v0e0/lxml_77372c3a77d843208173a320a0f71d8f/setup.py'"'"'; __file__='"'"'/tmp/pip-install-e6s2v0e0/lxml_77372c3a77d843208173a320a0f71d8f/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-e9czxmth
         cwd: /tmp/pip-install-e6s2v0e0/lxml_77372c3a77d843208173a320a0f71d8f/
    Complete output (3 lines):
    Building lxml version 4.6.0.
    Building without Cython.
    Error: Please make sure the libxml2 and libxslt development packages are installed.
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/3c/5a/81446fb5a4706bb07a98427634ab93219dc7fa393841d81f5da5c7e151b7/lxml-4.6.0.tar.gz#sha256=8319abacbbe9fd9791c2e3c23deb9b6a16abc0fcbdfcc26722035e865b3d207f (from https://pypi.org/simple/lxml/) (requires-python:>=2.7, !=3.0.*, !=3.1.*, !=3.2.*, !=3.3.*, != 3.4.*). Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
  Using cached lxml-4.5.2.tar.gz (4.5 MB)
    ERROR: Command errored out with exit status 1:
     command: /home/pi/.pyenv/versions/3.9.2/bin/python3.9 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-e6s2v0e0/lxml_8c4433996f964ed3815150c041dc92dc/setup.py'"'"'; __file__='"'"'/tmp/pip-install-e6s2v0e0/lxml_8c4433996f964ed3815150c041dc92dc/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-pip-egg-info-97t2rkmk
         cwd: /tmp/pip-install-e6s2v0e0/lxml_8c4433996f964ed3815150c041dc92dc/
    Complete output (3 lines):
    Building lxml version 4.5.2.
    Building without Cython.
    Error: Please make sure the libxml2 and libxslt development packages are installed.
    ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/2c/4d/3ec1ea8512a7fbf57f02dee3035e2cce2d63d0e9c0ab8e4e376e01452597/lxml-4.5.2.tar.gz#sha256=cdc13a1682b2a6241080745b1953719e7fe0850b40a5c71ca574f090a1391df6 ( [... cut ...] '"'; __file__='"'"'/tmp/pip-install-e6s2v0e0/lxml_e5374d7bea8342ab818cd8729f003bd9/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-cl_71_x4/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/.pyenv/versions/3.9.2/include/python3.9/lxml Check the logs for full command output.
pi@raspberrypi:~ $ 
```

AndKe, 2021-03-18:
Update: I got it installed by ```sudo apt-get install libxml2-dev libxslt-dev python-dev```
as per instructions: https://lxml.de/installation.html
With prerequisites like that, a bit of text-help would be nice.