ArduPilot/pymavlink #492: Could not connect to external PX4 SITL via ethernet

copterspace, 2020-12-16:
Dear colleagues, I have PX4 SITL [Jmavsim](https://dev.px4.io/v1.11/en/simulation/jmavsim.html) running on Ubuntu computer, ip address is 192.168.1.17
I try to connect to this SITL from Raspberry PI onboard computer via ethernet.
[Mavros](http://wiki.ros.org/mavros) connection with `roslaunch mavros px4.launch fcu_url:="udp://@192.168.1.17:14580"` works OK - I get valid mavlink heartbit with `rostopic echo /mavros/state`

But Python 3.7 `the_connection = mavutil.mavlink_connection('udpin:192.168.1.17:14580')` thows exception:

```
Traceback (most recent call last):
  File "test_pymavlink.py", line 23, in <module>
    the_connection = mavutil.mavlink_connection('udpin:192.168.1.17:14580')
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1680, in mavlink_connection
    return mavudp(device[6:], input=True, source_system=source_system, source_component=source_component, use_native=use_native)
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1038, in __init__
    self.port.bind((a[0], int(a[1])))
OSError: [Errno 99] Cannot assign requested address
```
If I try `the_connection = mavutil.mavlink_connection('udpout:192.168.1.17:14580')` - it does not throw exception, but then `the_connection.wait_heartbeat()` waits endlessly.

I could not find any example of connection to external SITL ip - I searched in [ArduSub book](https://www.ardusub.com/developers/pymavlink.html) and in [examples](https://github.com/ArduPilot/pymavlink/tree/master/examples).
Should this connection work? 
<br>How to establish connection to **external** SITL via ethernet and to receive Mavlink messages properly?

copterspace, 2020-12-17:
Tried tcp connection: `the_connection = mavutil.mavlink_connection('tcp:192.168.1.17:14580', autoreconnect=True, retries=10)`
but get "Connection refused" error:
```
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
[Errno 111] Connection refused sleeping
Traceback (most recent call last):
  File "test_pymavlink.py", line 26, in <module>
    retries=10) #, dialect = 'standard')
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1676, in mavlink_connection
    use_native=use_native)
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1192, in __init__
    self.do_connect()
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1212, in do_connect
    raise e
  File "/home/pi/.local/lib/python3.7/site-packages/pymavlink/mavutil.py", line 1205, in do_connect
    self.port.connect(self.destination_addr)
ConnectionRefusedError: [Errno 111] Connection refused
```
Although mavros connects OK to the same SITL. Mavlink status is:
```
instance #0:
	mavlink chan: #0
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 12.822 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Normal
	MAVLink version: 1
	transport protocol: UDP (18570, remote port: 14550)

instance #1:
	mavlink chan: #1
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 14.407 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 1.000
	  tx rate max: 4000000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14580, remote port: 14540)

instance #2:
	mavlink chan: #2
	type:		GENERIC LINK OR RADIO
	flow control: OFF
	rates:
	  tx: 2.875 kB/s
	  txerr: 0.000 kB/s
	  tx rate mult: 0.141
	  tx rate max: 4000 B/s
	  rx: 0.000 kB/s
	FTP enabled: YES, TX enabled: YES
	mode: Onboard
	MAVLink version: 1
	transport protocol: UDP (14280, remote port: 14030)
```

Chandler-Kluser, 2025-05-20:
Did someone managed to make it work? I have the same issue here