ArduPilot/pymavlink #496: Message Definitions in PyPi Release very outdated (more than 4 years old) ?!

zzaurak, 2021-01-05:
Hi, can you please bump the message definitions in the PyPi release to something newer than 4 years old? I was trying to implement VIDEO_STREAM_INFORMATION request and it is not included in the PyPI included message definitions, but was committed to mavlink/mavlink about 4 years ago (https://github.com/mavlink/mavlink/commit/bebff3ed976ee79668a6f2312132be3c6ca303cd) - there have been multiple releases of https://github.com/mavlink/mavlink since then. @tridge @jgoppert

amilcarlucas, 2021-01-07:
The message definitions are newer that that. I am the person that makes sure they get updated.
Only messages marked as WIP are not updated, and some other small exceptions. All of the exceptions have a reason. Nobody is being lazy here.
Please do a file by file compare between repos and you will understand what I mean.

But yes, if you do find something that you want to improve on, do a PR to the ArduPilot/mavlink downstream repository.

Closing this one now.

closed 2021-01-07 by amilcarlucas

zzaurak, 2021-01-07:
@adegroote Sorry, didn't mean to imply that I think anybody here would be lazy. I'm sure there are reasons for those diversions. Is there a reason the messages for the camera protocol are not yet supported (despite not being WIP anymore) ? If not, I will do a PR.

khancyr, 2021-01-07:
hello,

I am pretty sure that the difference come from the fact that Pymavlink is an ArduPilot software so release on PIP are done with ArduPilot mavlink messages that are differents from the main mavlink messages.

You can still update your pymavlink installation by manually installing it and providing it the PATH to the message definition you want to use. See in the readme for the right commands

zzaurak, 2021-01-07:
thanks, yeah i figured that out, too. I am using the mavlink/mavlink message definitions now with my manual install... still i think it is a bit cumbersome, that the pymavlink pip package is the "official" python bindings package, but really it is more like pymavlink-ardupilot package, as there is really a lot of diversion between the two message definition repos.

anyways thats more an issue i should raise at the mavlink/mavlink repo, maybe there can be an "official" pip package in the future.

I just raised this issue because I was frustrated that there was no straight forward way to implement and use the newer (but not WIP) messages besides a manual installation, but I guess I'm just new to the developer side of this project and thats just the state of it. 

Anyways, still, if you guys think it would be beneficial to keep this message definition fork as close as possible to the current releases (excluding WIP, i get that), and you'd like to have another contributor, I can do a PR. Let me know :)