ArduPilot/pymavlink #513: Can not receive message ATTITUDE_QUATERNION

kyrlon, 2021-03-02:
Currently I am attempting to acquire some general system information from an aircraft. Here is my script to acquire GPS and ATTITUDE from the system:

```
from pymavlink import mavutil
import time

the_connection = mavutil.mavlink_connection("/dev/ttyACM0", baud=115200)
# Request all data streams
data_rate = 1 # Hz
the_connection.mav.request_data_stream_send(the_connection.target_system, the_connection.target_component,
                                        mavutil.mavlink.MAV_DATA_STREAM_ALL, data_rate, 1)

gps_msg = "GLOBAL_POSITION_INT"
#message_type = "GPS_RAW_INT"
message_type = "ATTITUDE_QUATERNION"
attitude_msg = "ATTITUDE"

while True:
    try:
        gps_uav = the_connection.recv_match(type=gps_msg,blocking=False).to_dict()
        att_uav = the_connection.recv_match(type=attitude_msg,blocking=False).to_dict()
        print(att_uav)
        print(gps_uav)
    except:
        pass
    time.sleep(0.1)

```
And the following is the output:

```
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 5042451, 'lat': 339106096, 'lon': -845261530, 'alt': 317210, 'relative_alt': -4138, 'vx': 5, 'vy': 2, 'vz': 1, 'hdg': 26468}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 5042499, 'roll': -0.005039728246629238, 'pitch': 0.0005145773757249117, 'yaw': -1.663695216178894, 'rollspeed': 0.00011657504364848137, 'pitchspeed': 8.563127630623057e-05, 'yawspeed': 0.0001750187948346138}
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 5042481, 'lat': 339106096, 'lon': -845261529, 'alt': 317210, 'relative_alt': -4138, 'vx': 5, 'vy': 2, 'vz': 1, 'hdg': 26468}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 5042523, 'roll': -0.005051034037023783, 'pitch': 0.0005108385812491179, 'yaw': -1.6637033224105835, 'rollspeed': 0.00036399089731276035, 'pitchspeed': -0.00024093712272588164, 'yawspeed': -0.00043800967978313565}
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 5042499, 'lat': 339106096, 'lon': -845261529, 'alt': 317210, 'relative_alt': -4138, 'vx': 5, 'vy': 2, 'vz': 1, 'hdg': 26468}
{'mavpackettype': 'ATTITUDE', 'time_boot_ms': 5043153, 'roll': -0.005177208222448826, 'pitch': 0.0008976433891803026, 'yaw': -1.6636784076690674, 'rollspeed': 0.00019293429795652628, 'pitchspeed': 0.0008838956127874553, 'yawspeed': -0.00017069702153094113}
{'mavpackettype': 'GLOBAL_POSITION_INT', 'time_boot_ms': 5042526, 'lat': 339106096, 'lon': -845261529, 'alt': 317210, 'relative_alt': -4137, 'vx': 5, 'vy': 2, 'vz': 1, 'hdg': 26468}
```

When I attempt to change the message type to ATTITUDE_QUATERNION, I get no response, or rather the script hangs and no messages are populated. I am not sure why this specific message is not shown. 

My question is, do I need to enable this specific system information for this aircraft in the mission planner?

kyrlon, 2021-03-02:
It seems this problem has been present in a dronekit [issue](https://github.com/dronekit/dronekit-python/issues/987)

kyrlon, 2021-03-03:
Hopefully this is resolved soon. Here is a PR [request](https://github.com/ArduPilot/ardupilot/pull/16798) pending for Ardupilot.

peterbarker, 2022-01-12:
@kyrlon well, ArduPilot can now emit `ATTITUDE_QUATERNION` - so this should be fixed.  Please re-open if there's still an issue.

closed 2022-01-12 by peterbarker