ArduPilot/pymavlink #517: PX4 mission upload fail

BlackJocker1995, 2021-03-10:
I write a program to upload a WPL 110 mission to drone. When the firmware is ArduPilot simulator, it accepts mission. But when the firmware is PX4 simulator, it can not upload successfully.
It repeat sending `MISSION_REQUEST {target_system : 255, target_component : 0, seq : 0, mission_type : 0}`, but not the next seq.
And the terminal shows `INFO  [tone_alarm] notify negative`.
I just wonder they are incompatible?

```python
loader = mavwp.MAVWPLoader()
loader.load(mission_file)
logging.debug(f"Load mission file {mission_file}")
self._master.waypoint_clear_all_send()
self._master.waypoint_count_send(loader.count())
seq_list = [True] * loader.count()
while True in seq_list:
    msg = self._master.recv_match(type=['MISSION_REQUEST', 'MISSION_REQUEST_INT'], blocking=True, timeout=timeout)
    if msg is not None and seq_list[msg.seq] is True:              
        self._master.mav.send(loader.wp(msg.seq))
        seq_list[msg.seq] = False
        logging.debug('Sending waypoint {0}'.format(msg.seq))
mission_ack_msg = self._master.recv_match(type=['MISSION_ACK'], blocking=True, timeout=timeout)
```

peterbarker, 2021-03-11:
On Tue, 9 Mar 2021, BlackJocker1995 wrote:

> I just wonder they are incompatible?

Yes.  ArduPilot stores the dynamically-updating home location in seq=0 - 
PX4 Firmware does not.  It sucks.

closed 2021-03-11 by BlackJocker1995

Czw007, 2022-04-19:
> I write a program to upload a WPL 110 mission to drone. When the firmware is ArduPilot simulator, drone accepts mission. But when the firmware is PX4 simulator, it can not upload successfully. It repeat receiving `MISSION_REQUEST {target_system : 255, target_component : 0, seq : 0, mission_type : 0}` from drone, but not the next seq. And the terminal shows `INFO [tone_alarm] notify negative`. I just wonder they are incompatible?
> 
> ```python
> loader = mavwp.MAVWPLoader()
> loader.load(mission_file)
> logging.debug(f"Load mission file {mission_file}")
> self._master.waypoint_clear_all_send()
> self._master.waypoint_count_send(loader.count())
> seq_list = [True] * loader.count()
> while True in seq_list:
>     msg = self._master.recv_match(type=['MISSION_REQUEST', 'MISSION_REQUEST_INT'], blocking=True, timeout=timeout)
>     if msg is not None and seq_list[msg.seq] is True:              
>         self._master.mav.send(loader.wp(msg.seq))
>         seq_list[msg.seq] = False
>         logging.debug('Sending waypoint {0}'.format(msg.seq))
> mission_ack_msg = self._master.recv_match(type=['MISSION_ACK'], blocking=True, timeout=timeout)
> ```
I recently reproduced your article 2022-ICSE-LGdFuzzer normally, and I also encountered the same problem. May I ask how this problem was finally solved.

BlackJocker1995, 2022-04-19:
> I recently reproduced your article 2022-ICSE-LGdFuzzer normally, and I also encountered the same problem. May I ask how this problem was finally solved.

In our experiment, only Arduiplot was finally selected, and PX4 was not tested. There are differences in the configuration of the two.

Czw007, 2022-04-19:
> In our experiment, only Arduiplot was finally selected, and PX4 was not tested. There are differences in the configuration of the two.

Thank you very much for your reply. In your paper, you also did a comparative experiment with the 2019-Usenix-RVFuzzer paper. As far as I know, RVFuzzer is not open source. Do you reproduce it yourself? , or obtained the source code from the original author.