ArduPilot/pymavlink #521: ModuleNotFoundError: No module named 'pymavlink'

kucar17, 2021-03-25:
I have just installed SITL and MAVProxy. I cloned the repo and updated it with the --recursive parameter and etc.

When I use the command
`cd ~/ardupilot/ArduCopter
../Tools/autotest/sim_vehicle.py --map --console`

I get "ModuleNotFoundError: No module named 'pymavlink'" error. I tried `pip install pymavlink,` and I tried `pip install "pymavlink==2.4.6"`. The result is the same.

I went to the autotest directory and tried `python sim_vehicle.py` or `python3 sim_vehicle.py` and The result has not changed yet again.

When I tried `pip install pymavlink` I received the following error:

`Collecting pymavlink
  Using cached pymavlink-2.4.14.tar.gz (4.1 MB)
Requirement already satisfied: future in /usr/lib/python3.7/site-packages (from pymavlink) (0.18.2)
Collecting lxml
  Using cached lxml-4.6.3.tar.gz (3.2 MB)
Using legacy 'setup.py install' for pymavlink, since package 'wheel' is not installed.
Using legacy 'setup.py install' for lxml, since package 'wheel' is not installed.
Installing collected packages: lxml, pymavlink
    Running setup.py install for lxml ... error
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-32g5fsep/lxml_317de54fd3ea47398f2612592d55dd1a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-32g5fsep/lxml_317de54fd3ea47398f2612592d55dd1a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-x0qddohp/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.7m/lxml
         cwd: /tmp/pip-install-32g5fsep/lxml_317de54fd3ea47398f2612592d55dd1a/
    Complete output (86 lines):
    Building lxml version 4.6.3.
    Building without Cython.
    Building against libxml2 2.9.10 and libxslt 1.1.29
    running install
    running build
    running build_py
    creating build
    creating build/lib.cygwin-3.1.7-x86_64-3.7
    creating build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/builder.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/cssselect.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/doctestcompare.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/ElementInclude.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/pyclasslookup.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/sax.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/usedoctest.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/_elementpath.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/__init__.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    creating build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    copying src/lxml/includes/__init__.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    creating build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/builder.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/clean.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/defs.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/diff.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/ElementSoup.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/formfill.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/html5parser.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/soupparser.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/usedoctest.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/_diffcommand.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/_html5builder.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/_setmixin.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    copying src/lxml/html/__init__.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/html
    creating build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron
    copying src/lxml/isoschematron/__init__.py -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron
    copying src/lxml/etree.h -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/etree_api.h -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/lxml.etree.h -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/lxml.etree_api.h -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml
    copying src/lxml/includes/c14n.pxd -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    copying src/lxml/includes/config.pxd -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    copying src/lxml/includes/dtdvalid.pxd -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    copying src/lxml/includes/etreepublic.pxd -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
    copying src/lxml/includes/htmlparser.pxd -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/includes
[... 21 lines cut ...]
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_dsdl_include.xsl -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_message.xsl -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_schematron_skeleton_for_xslt1.xsl -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/iso_svrl_for_xslt1.xsl -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    copying src/lxml/isoschematron/resources/xsl/iso-schematron-xslt1/readme.txt -> build/lib.cygwin-3.1.7-x86_64-3.7/lxml/isoschematron/resources/xsl/iso-schematron-xslt1
    running build_ext
    building 'lxml.etree' extension
    creating build/temp.cygwin-3.1.7-x86_64-3.7
    creating build/temp.cygwin-3.1.7-x86_64-3.7/src
    creating build/temp.cygwin-3.1.7-x86_64-3.7/src/lxml
    gcc -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -ggdb -O2 -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong --param=ssp-buffer-size=4 -fdebug-prefix-map=/pub/devel/python/python37/python37-3.7.10-1.x86_64/build=/usr/src/debug/python37-3.7.10-1 -fdebug-prefix-map=/pub/devel/python/python37/python37-3.7.10-1.x86_64/src/Python-3.7.10=/usr/src/debug/python37-3.7.10-1 -ggdb -O2 -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong --param=ssp-buffer-size=4 -fdebug-prefix-map=/pub/devel/python/python37/python37-3.7.10-1.x86_64/build=/usr/src/debug/python37-3.7.10-1 -fdebug-prefix-map=/pub/devel/python/python37/python37-3.7.10-1.x86_64/src/Python-3.7.10=/usr/src/debug/python37-3.7.10-1 -DCYTHON_CLINE_IN_TRACEBACK=0 -I/usr/include/libxml2 -Isrc -Isrc/lxml/includes -I/usr/include/python3.7m -c src/lxml/etree.c -o build/temp.cygwin-3.1.7-x86_64-3.7/src/lxml/etree.o -w
    src/lxml/etree.c:97:10: fatal error: Python.h: No such file or directory
       97 | #include "Python.h"
          |          ^~~~~~~~~~
    compilation terminated.
    Compile failed: command 'gcc' failed with exit status 1
    creating tmp
    cc -I/usr/include/libxml2 -I/usr/include/libxml2 -c /tmp/xmlXPathInitpjodw_ei.c -o tmp/xmlXPathInitpjodw_ei.o
    cc tmp/xmlXPathInitpjodw_ei.o -lxml2 -o a.out.exe
    error: command 'gcc' failed with exit status 1
    ----------------------------------------
ERROR: Command errored out with exit status 1: /usr/bin/python3.7 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-32g5fsep/lxml_317de54fd3ea47398f2612592d55dd1a/setup.py'"'"'; __file__='"'"'/tmp/pip-install-32g5fsep/lxml_317de54fd3ea47398f2612592d55dd1a/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-x0qddohp/install-record.txt --single-version-externally-managed --compile --install-headers /usr/include/python3.7m/lxml Check the logs for full command output.
`

Can anybody help, I am one week away from a project and I do not want to lose time in the beginnig parts.

khancyr, 2021-03-25:
Hello,

Install python-dev with cygwin and it should solve the issue

auturgy, 2021-09-01:
Closing as resolved.  If this issue is not resolved, feel free to re-open or ask on the forums/discord

closed 2021-09-01 by auturgy