ArduPilot/pymavlink #522: common.py has MAV_CMD_ACK_OK = 1, but should be MAV_CMD_ACK_OK = 0

PavloGV, 2021-04-01:
My apologies if I've missed something, however I noticed that if one installs pymavlink via command line, the resulting common.py has MAV_CMD_ACK_OK = 1, but in the official mavlink repo, their common.xml has MAV_CMD_ACK_OK = 0. 

A bit more background in case it's useful: I was transmitting from my PC to a microcontroller using the c-generated mavlink from the official mavlink mavgenerate.py. I had used the common.xml from their message_definitiions/v1.0/, but when I sent a MAV_CMD_ACK_OK from the PC using pymavlink I realized that the micro was reading a MAV_CMD_ACK_ERR_FAIL, which has a value of 1 in the official mavlink common.h/common.xml. 

My work around for now, was I re-generated the c-generated mavlink with the same mavgenerate.py as before, but chose the common.xml from pymavlink. 

Shouldn't they both have MAV_CMD_ACK_OK = 0. It should match right? I understand you can have different dialects, but common should be the same everywhere, right? I mean it is called common after all. Again, I apologize in advance if there is a nuance here I haven't learned yet, but from a practical standpoint and to make this a bit easier for people new to this, MAV_CMD_ACK_OK = 0 everywhere.  

This last point is not as important, but rather just slightly to my dismay; My current solution that I mentioned before is that both the micro and the PC "agree" that MAV_CMD_ACK_OK = 1, but I'd prefer that it be zero because, let's face it, who are we, MATLAB? I have to poke a little bit of fun at this. 

Peace out!

peterbarker, 2021-04-02:
On Wed, 31 Mar 2021, PavloGV wrote:

> My apologies if I've missed something, however I noticed that if one installs pymavlink via command line, the resulting common.py has MAV_CMD_ACK_OK = 1, but in the official mavlink repo,
> their common.xml has MAV_CMD_ACK_OK = 0.

Definitely should be 0 - this seems to be a pretty significant bug.

Generating locally here using

pushd ~/rc/pymavlink
MDEF=$HOME/rc/ardupilot/modules/mavlink/message_definitions python 
setup.py build install --user --force

does not seem to replicate, but I can see vestiges of a slugs.xml which 
did have this problem.

Steps to reproduce?

olliw42, 2021-04-26:
https://github.com/mavlink/mavlink/pull/1592

peterbarker, 2021-07-02:
> [mavlink/mavlink#1592](https://github.com/mavlink/mavlink/pull/1592)

Thanks @olliw42 - closing this as a fix seems to have been merged.

closed 2021-07-02 by peterbarker