ArduPilot/pymavlink #532: enum_is_a_bitmask does not work correctly

python36, 2021-04-19:
https://github.com/ArduPilot/pymavlink/blob/4dac715e4e394714d0676fa6675ff89d2871f380/generator/mavgen_swift.py#L249-L257
This don't work for https://mavlink.io/en/messages/common.html#ATTITUDE_TARGET_TYPEMASK

shancock884, 2024-01-11:
If an enumeration is a bitmask, it is defined as such in the XML specification, so I think that this function is redundant, and [line 302 of mavgen_swift.py](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavgen_swift.py#L302) could simply become:
```python
enum.is_a_bitmask = enum.bitmask
```

Abtektas, 2026-10-01:
This is still the case on current master (c6b7a51). With `common.xml` from mavlink master (527637cb3), `enum_is_a_bitmask()` disagrees with the `bitmask` attribute of the XML for 7 of the 152 enums:

| Enum | `bitmask` in XML | `enum_is_a_bitmask()` |
|---|---|---|
| `ATTITUDE_TARGET_TYPEMASK` | true | false |
| `ADSB_FLAGS` | true | false |
| `GIMBAL_DEVICE_CAP_FLAGS` | true | false |
| `GIMBAL_MANAGER_CAP_FLAGS` | true | false |
| `CAMERA_TRACKING_STATUS_FLAGS` | true | false |
| `HIL_SENSOR_UPDATED_FLAGS` | true | false |
| `TUNE_FORMAT` | false | true |

The first six are generated as plain enums instead of option sets, because the heuristic only accepts the values 1, 2, 4, 8, ... without gaps (`ATTITUDE_TARGET_TYPEMASK` has 1, 2, 4, 32, 64, 128, and `CAMERA_TRACKING_STATUS_FLAGS` also has 0). `TUNE_FORMAT` only has the values 1 and 2, so it becomes an option set although it isn't a bitmask.

As @shancock884 suggested, using `enum.bitmask` from the XML instead of the heuristic fixes all seven. I'd like to open a PR for this, and check that the generated Swift code still compiles.

*Assisted-by: Claude:claude-opus-5-5*