ArduPilot/pymavlink #560: Generate MAVLink Messages to Trick QCG

natetoombs, 2021-06-18:
I'm trying to get an autopilot that uses ROS messages to communicate with QGroundControl, but I am having trouble translating the ROS messages to MAVLink messages and broadcasting them.

First, I just want to run a script that will publish sufficient MAVLink messages to connect with QGC as a drone. How can I use pymavlink to give QGC all of the messages it requires to think that there is a drone connected and flying?

khancyr, 2021-06-18:
Hello,
If you need a ROS-mavlink bridge, look at mavros

khancyr, 2021-06-18:
More generally, you just need to implement the Heartbeat message to be registered o  a mavlink network.

jmachuca77, 2021-06-18:
This is more of a support question, I think you will get more help posting on the discord mavlink or mavproxy channels, Try posting the question there. I will close this issue.

closed 2021-06-18 by jmachuca77