ArduPilot/pymavlink #566: pyinstaller xml Error.....

KimJinHyeok1, 2021-06-23:
hello now I'm develop the GCS program using dronekit and PyQT5, pyinstaller

in the pycharm everything work well!

so i make .exe file using Pyinstaller

but when i click Connect button (vehicle=connect(udpin:localhost:port))

In the console print "warning: unable to load xml validator libraries. xml validation will not be performed"

I'm doing almost solution in the internet ( ex: hidden import lxml, lxml.etree...., reinstall pyinstaller,pymavlink,mavutil...)
but error is same..

I'm really don't know why that error occur only in the .exe file..

Pleeeeeeeeeease. Let me know any solution..... 

Thankyou

peterbarker, 2021-07-02:
On Wed, 23 Jun 2021, KimJinHyeok1 wrote:

> In the console print "warning: unable to load xml validator libraries. xml validation will not be performed"

Missing lxml Python package.

closed 2021-07-02 by peterbarker

gregor-coder, 2026-10-28:
This warning is caused by a missing mavschema.xsd resource file in your PyInstaller distribution. pymavlink uses this XML Schema Definition (XSD) file to validate MAVLink XML dialects.

Add `--collect-data pymavlink` to pyinstaller command line.

This tells PyInstaller to collect package data files from pymavlink, including mavschema.xsd when it is present in the installed package.