ArduPilot/pymavlink #571: Problem with latest pymavlink - mavutil

mtbsteve, 2021-07-14:
When I import pymavlink with `from pymavlink import mavutil`
I am getting the following error:
```
Traceback (most recent call last):
  File "/home/apsync/ws/src/zed_yolo/nodes/zed_ros_to_mavlink.py", line 34, in <module>
    from pymavlink import mavutil
  File "/home/apsync/.local/lib/python3.7/site-packages/pymavlink-2.4.15-py3.7-linux-aarch64.egg/pymavlink/mavutil.py", line 2005, in <module>
    mavlink.MAV_TYPE_DECAROTOR:   mode_mapping_acm,
AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_TYPE_DECAROTOR'
```
I cloned the repro, and built it with `python3 setup.py install --user` from within the pymavlink directory. No errors or warnings during build.
Any idea anyone? Thanks!

peterbarker, 2021-07-14:
On Wed, 14 Jul 2021, mtbsteve wrote:

> AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_TYPE_DECAROTOR'

Update your mavlink definitions.  If that's not a possiblity, we can 
potentially work around this.

peterbarker, 2021-09-20:
@mtbsteve did you get this fixed?

mtbsteve, 2021-09-20:
Yes I managed to get it working- thanks!

closed 2021-09-20 by mtbsteve

tmendoza271, 2023-09-21:
> On Wed, 14 Jul 2021, mtbsteve wrote: AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_TYPE_DECAROTOR'
> Update your mavlink definitions. If that's not a possiblity, we can potentially work around this.

What do you mean by "update your mavlink definitions"? I'm having the same issue

marko-castral, 2023-10-09:
> > On Wed, 14 Jul 2021, mtbsteve wrote: AttributeError: module 'pymavlink.dialects.v20.ardupilotmega' has no attribute 'MAV_TYPE_DECAROTOR'
> > Update your mavlink definitions. If that's not a possiblity, we can potentially work around this.
> 
> @peterbarker What do you mean by "update your mavlink definitions"? I'm having the same issue

Looks like this pops on different places. Same issues with different dialect (v10.ardupilotmega) here. TMAV_TYPE_VTOL_(TAILSITTER)_QUADROTOR and MAV_TYPE_VTOL_(TAILSITTER)_DUALROTOR have to be renamed accordingly in your lib scripts.