ArduPilot/pymavlink #595: pymavlink not able to encode a message after receiving

dayjaby, 2021-11-24:
Let's say you receive a message with `d = self.mavconn.recv_match()`, e.g. a common.GIMBAL_DEVICE_INFORMATION. You end up with an instance of the class `pymavlink.dialects.v20.common.MAVLink_gimbal_device_information_message` with all `char[??]` fields being automatically decoded as string. So `d.vendor_name` is of type `str`.

However, if you pack this MAVLink_gimbal_device_information_message, by calling its class method `pack` with the mav object `self.mavconn.mav`, which is expected to be fine ... like calling this method with the same mav object that was also used to decode it, you end up getting an error:

```
  File "main.py", line 24, in loop
    msgbuf = d.pack(self.mavconn.mav)
  File "/usr/local/lib/python3.8/dist-packages/pymavlink-2.4.15-py3.8-linux-x86_64.egg/pymavlink/dialects/v20/common.py", line 12451, in pack
    return MAVLink_message.pack(self, mav, 74, struct.pack('<QIIIffffffHH32s32s32s', self.uid, self.time_boot_ms, self.firmware_version, self.hardware_version, self.roll_min, self.roll_max, self.pitch_min, self.pitch_max, self.yaw_min, self.yaw_max, self.cap_flags, self.custom_cap_flags, self.vendor_name, self.model_name, self.custom_name), force_mavlink1=force_mavlink1)
struct.error: argument for 's' must be a bytes object
```

I guess you have to decide whether the message fields have to be EITHER strings OR bytes.

mhcekic22, 2021-12-03:
I faced the same error, Did you figure it out the problem?

alehed, 2022-07-15:
This problem should be fixed by #666 which improves the string/bytes handling.

dayjaby, 2022-07-15:
@alehed Very much appreciated! :)

closed 2022-07-15 by dayjaby