ArduPilot/pymavlink #596: Generating dialects in arbitrary directory on the fly

basil-huber, 2021-11-28:
Hi folks,

As I understand it, custom dialects can be generated on the fly when creating a connection with `mavlink_connection(..., dialect=mydialect)`. However, this requires the definitions to be just next to the `pymavlink` directory, which is quite a dark place when installing pymavlink through pip.

As an example, I have a python package which depends on pymavlink and used a custom dialect. It would be really nice to be able to simply put `install_requires=['pymavlink]` and then call
```
os.environ['MDEF'] = r'path/to/my/definitions'
mavlink_conn = mavutil.mavlink_connection(..., dialect='mydialect')
```

Cheers,
Basil

basil-huber, 2021-11-28:
I just discovered that I can achieve the same with `os.environ['MAVLINK_DIALECT'] = path/to/my/dir`.

peterbarker, 2022-01-13:
@basil-huber I've created the link PR to attempt to fix this one.  Works in my local testing.

Your XML directory does need to be structured to include the "v1.0" for this to work.

closed 2024-04-03 by peterbarker