ArduPilot/pymavlink #602: mavlink_types.h:92: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]

AleksandrBon, 2021-12-02:
On g++ compilation with "-pedantic" flag error:
mavlink_types.h:92: error: ISO C++ prohibits anonymous structs [-Werror=pedantic]
};
^
mavlink_types.h:17: in definition of macro ‘MAVPACKED’
#define MAVPACKED( Declaration ) Declaration attribute((packed))
^~~~~~~~~~~~~~~
etc.

This problem is originated in that the c++ standard forbids anonymous structs declarations
https://stackoverflow.com/questions/2253878/why-does-c-disallow-anonymous-structs

And even if mavgenerate.py one can choose to generate c++ headers, core files like mavlink_types.h stays same.

peterbarker, 2021-12-02:
Fix here is probably to just specify the minimum standard we're allowing.  Brief research tells me that C++-11 introduced anonymous structs.

Which compiler were you using that decided the structs were bad?  What standard was it enforcing, and why?

You've managed to pack three separate issues into here... that last one seems to be the most significant.  PR or more information on that one (in a separate issue referencing this one) would be great!

spiderkeys, 2021-12-04:
I get the `mavlink_types.h:92:5: warning: ISO C++ prohibits anonymous structs [-Wpedantic]` warning using:

g++ (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0

CMake options:
```cmake
target_compile_options(app 
PUBLIC
  -Wall 
  -Wextra 
  -Wpedantic
)

target_compile_features(app
PUBLIC
  cxx_std_17
)
```

AleksandrBon, 2021-12-05:
> Fix here is probably to just specify the minimum standard we're allowing. Brief research tells me that C++-11 introduced anonymous structs.
> 
> Which compiler were you using that decided the structs were bad? What standard was it enforcing, and why?
> 

compiler: gnu gcc 7.5.0 on Ubuntu 18.04
compiler flags:
-std=c++11
-Werror
-Wall
-Wextra
-pedantic

Of course as mentioned in reply from @spiderkeys if flag **Werror** is not set compiler will produce warning and not an error.

peterbarker, 2022-03-24:
@AleksandrBon @spiderkeys so a little further reading tells me that C11 includes anon structs but C++11 does not.

So I guess we should probably name these structs at some stage (or make the C++ bindings better....)

In the meantime, perhaps add `-fms-extensions` to avoid the warning/error.

AleksandrBon, 2022-03-30:
Ok. I will try.

closed 2022-03-30 by AleksandrBon