ArduPilot/pymavlink #607: c MAVLINK_HELPER function mavlink_get_message_info_by_name never finds index 0 in MAVLINK_MESSAGE_NAMES

dlwalter, 2021-12-10:
Ran into an issue using the mavlink_get_message_info_by_name() and it would not find the mavlink message `ACTUATOR_CONTROL_TARGETS`, which is the first entry in my dialect's MAVLINK_MESSAGE_NAMES.

https://github.com/ArduPilot/pymavlink/blob/4dac715e4e394714d0676fa6675ff89d2871f380/generator/C/include_v2.0/mavlink_get_info.h#L47

Looking at the search algorithm, mid will never reach index 0 to run strcmp() on the message name.

An example dialect has `# define MAVLINK_MESSAGE_NAMES {{ "ACTUATOR_CONTROL_TARGET", 140 }, { "ADSB_VEHICLE", 246 }, ...`

At the end of the search, `low=0`, `high=1`, `mid=1`

`strcmp()` will return >0 after comparing `"ADSB_VEHICLE"` to `"ACTUATOR_CONTROL_TARGET"`

then `high=0` and `mid=1` and the search will end.

dlwalter, 2021-12-10:
It looks like a possible fix would be line 55

https://github.com/ArduPilot/pymavlink/blob/4dac715e4e394714d0676fa6675ff89d2871f380/generator/C/include_v2.0/mavlink_get_info.h#L55

changing the while loop to `while (low <= high)` which would run a check with `mid=0` when `low=0` and `high=0`

peterbarker, 2022-01-12:
@dlwalter thanks for the bug report - I was able to replicate relatively easily.

I've created https://github.com/ArduPilot/pymavlink/pull/613 to try to address the issue - are you able to test?

closed 2022-01-12 by tridge