ArduPilot/pymavlink #610: recv_match(blocking=False, timeout=1)  - blocks anyway

AndKe, 2021-12-29:
recv_match(blocking=False, timeout=1)
..simply does not work as expected.  It blocks anyway.

Please see thread: https://discuss.ardupilot.org/t/non-blocking-pymavlink-trouble-recv-match-blocking-false-timeout-1/70151

peterbarker, 2022-01-12:
I am unable to replicate this failure.

I'm using this script:
```
#!/usr/bin/env python

from pymavlink import mavutil

import optparse

class connect_and_dump(object):
    def __init__(self, connect_string):
        self.connect_string = connect_string

    def run(self):
        print("Connect (%s)" % self.connect_string)
        mav = mavutil.mavlink_connection(self.connect_string)

        while True:
            m = mav.recv_match(timeout=1)
            if m is None:
                print(".", end='')
                continue
            print("\n%s" % m)

if __name__ == '__main__':
    parser = optparse.OptionParser("connect_and_dump")
    opts, args = parser.parse_args()

    connect_string = args[0]

    c = connect_and_dump(connect_string)
    c.run()
```

And I've run it like this:

```
pbarker@bluebottle:~/rc$ ./connect_and_dump.py  tcpin:localhost:7654
pbarker@bluebottle:~/rc$ ./connect_and_dump.py  localhost:7654
```

I'm using MAVProxy with `output add` to get a mavlink stream on those ports.

When I stop the output stream (ctrl-z MAVProxy to pause it) I get vast numbers of "."s being printed out, indicating recv_match is always returning.

@AndKe Obviously my testing methodology is flawed - any hints?

AndKe, 2025-08-31:
@mujtabachang if your comment means that you still can replicate this issue, then say so, that would be valuable for the devs to know.

closed 2025-09-02 by khancyr