ArduPilot/pymavlink #615: Problem with generator for C (message without fields)

Kazeshirou, 2022-01-12:
comma before `${{arg_fields: ${array_const}${type} ${array_prefix}${name}, }}` breaks code for empty messages (without fields)

For example
https://github.com/ArduPilot/pymavlink/blob/8c72af35d782224b8886e8f64205ac358e2ea056/generator/mavgen_c.py#L358

This error appears in few places in file.

It's interesting that generating test code processes empty messages correctly  `${{arg_fields:, packet1.${name} }}`  

https://github.com/ArduPilot/pymavlink/blob/8c72af35d782224b8886e8f64205ac358e2ea056/generator/mavgen_c.py#L486

peterbarker, 2022-01-13:
@Kazeshirou is this an important thing to fix?

I wouldn't be surprised to find out a lot of the generators are broken.

Generally supporting empty messages can be a problem in some languages as zero-length objects can't have memory addresses (not a problem in our code, but perhaps someone else generating headers will have a problem) - so perhaps we should just say that a mavlink message has to have at least one field?

Kazeshirou, 2022-01-13:
"mavlink message has to have at least one field" warning in documentation is also good solution, why not. This fact has to be obvious from docs. Right now, it isn't

peterbarker, 2022-01-13:
Where would you expect to see this noted?

I love making things documentations problems (@hamishwillee ) :-)

Kazeshirou, 2022-01-13:
https://ardupilot.org/dev/docs/mavlink-basics.html
In Message definition, right after "Messages are no more than 263 bytes" =) 

And in readme in the end of section [Documention](https://github.com/ArduPilot/pymavlink#documentation)

Kazeshirou, 2022-01-13:
and  Data (1 - 255 byte not 0 ) in https://github.com/ArduPilot/ardupilot_wiki/blob/master/dev/source/docs/mavlink-basics.rst  that

hamishwillee, 2022-01-14:
@peterbarker If we agree the "messages must have at least one field rule", the best way to help users would be to add it to validation in https://github.com/ArduPilot/pymavlink/blob/master/generator/mavschema.xsd . @amilcarlucas Could you make that happen?
From my perspective this is fine, because it was a limitation I knew about "by osmosis". 

From my perspective that would get documented in https://mavlink.io/en/guide/xml_schema.html and https://mavlink.io/en/guide/define_xml_element.html

I'm off on holiday for two weeks, so won't get to this until I get back (and in any case would like to confirm the XSD gets updated first). But if I don't come back at that point feel free to ping me.

hamishwillee, 2022-02-03:
@amilcarlucas Can you make that update, or do you want me to ^^^ ?

amilcarlucas, 2022-02-03:
@hamishwillee can you do it?

hamishwillee, 2022-02-03:
Yes. Next week.

hamishwillee, 2022-02-08:
Fix created in #630 - I will roll to docs if that gets through. Note I have asked for confirmation that a broader group are happier with this. While and edge case, it feels kind of fundamental.

hamishwillee, 2022-02-16:
This is fixed and can now be closed - the validator checks for it and the docs about the XML schema note that at least one field is required.

peterbarker, 2022-02-16:
Thanks @hamishwillee

closed 2022-02-16 by peterbarker