ArduPilot/pymavlink #619: using command camera_information im getting an error when i send

mark-nick-o, 2022-01-13:
I'm trying to send the following message from the library 

Initially i thought it was the data type i sent to the function so i experimented with what i thought was a uint8_t[32] "array, numpy array list, string"

here is my function and here is the results it suggests to me that the function references an element outside the list/array/string

[image]

here is the code to my function which produces that output

    #
    # simple function to send the information about the camera from the robot
    #    
    def mavlink_send_camera_information(self, the_connection):

        vendor_name_nd = np.dtype([('ABB',np.uint8)])
        model_name_nd = np.dtype([('BAC',np.uint8)])
        vendor_name_list = [65,66,66] 
        model_name_list = [66,65,67]
        vendor_name = "ABB" 
        model_name = "BAC"
        #
        # convert string to ascii list and make numpy array
        #
        vn = []
        mn = []
        j = 0
        for j in range(len(model_name)):
            mn.append(ord(model_name[j]))
        k = 0
        for k in range(len(vendor_name)):
            vn.append(ord(vendor_name[k]))
        u8_model_name = np.array(mn, np.uint8)
        u8_vendor_name = np.array(vn, np.uint8)
        mn_u8 = u8_model_name.astype(np.uint8)
        vn_u8 = u8_vendor_name.astype(np.uint8)

        #
        # defined as arrays of bytes
        #
        arr_vendor = arr.array('B', [65,66,66])
        arr_model = arr.array('B', [66,65,67])
        arr2_vendor = arr.array('B', vn)
        arr2_model = arr.array('B', mn)
        print("\033[31m >>>>>>>>>>>> should be using 8bit byte array <<<<<<<<<<<<<<< ")
        print(f"\033[32m model {arr_model} {type(arr_model)} vendor {arr_vendor} {type(arr_vendor)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                arr_vendor, 
                arr_model, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(f" model {arr2_model} {type(arr2_model)} vendor {arr2_vendor} {type(arr2_vendor)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                arr2_vendor, 
                arr2_model, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(" \033[31m >>>>>>>>>>>>>>>>> using list <<<<<<<<<<<<<<<<<<<<<<< ")
        print(f" \033[34m model {model_name_list} {type(model_name_list)} {vendor_name_list} {type(vendor_name_list)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                vendor_name_list, 
                model_name_list, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(" \033[31m >>>>>>>>>>>>>>> numpy dtype <<<<<<<<<<<<<<<<<<<<<<< ")
        print(f"\033[33m model {vendor_name_nd} {type(vendor_name_nd)} {model_name_nd} {type(model_name_nd)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                vendor_name_nd, 
                model_name_nd, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(" \033[31m >>>>>>> using ord and numpy <<<<<<<< ")
        print(f"\033[36m model {vn} {type(vn)} {u8_model_name} vendor {mn} {type(mn)}")
        try:    
            the_connection.mav.camera_information_send(
                100,
                vn, 
                mn, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(f" vendor {u8_vendor_name} {type(u8_vendor_name)} model {u8_model_name} {type(u8_model_name)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                u8_vendor_name, 
                u8_model_name, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print(f" vendor {vn_u8} {type(vn_u8)} model {mn_u8} {type(mn_u8)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                vn_u8, 
                mn_u8, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        #exit(1)
        print(f" vendor {vendor_name} {type(vendor_name)} model {model_name} {type(model_name)} ")
        try:    
            the_connection.mav.camera_information_send(
                100,
                vendor_name, 
                model_name, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                0, 
                "a")
            ret = True
        except Exception as err_msg:
            print("Failed to send camera information message : %s" % (err_msg))
            ret = False
        print("\033[35m end>>>>>>>>>>>>>")
        return ret
        
    def mavlink_send_camera_settings(self, the_connection):
        #if self.mavlink10():
        try:    
            the_connection.mav.camera_settings_send(
                self.time_boot_ms,
                self.mode_id,                                                       #  Camera mode
                self.zoomLevel,                                                     #  Current zoom level (0.0 to 100.0, NaN if not known)*/
                self.focusLevel)  
            ret = True
        except Exception as err_msg:
            print("Failed to send camera settings message : %s" % (err_msg))
            ret = False
        return ret
        
    def mavlink_send_storage_information(self, the_connection):
        #if self.mavlink10():
        try:    
            the_connection.mav.storage_information_send(
                self.time_boot_ms, 
                self.storage_id, 
                self.storage_count, 
                self.status, 
                self.total_capacity, 
                self.used_capacity, 
                self.available_capacity, 
                self.read_speed, 
                self.write_speed)
            ret = True
        except Exception as err_msg:
            print("Failed to send storage information message : %s" % (err_msg))
            ret = False
        return ret

i then looked in my mavlink_libs i generated with mavgenerate and it looks like this array is not defined as per the spec order ??

[image]

amilcarlucas, 2022-01-20:
You need to follow the https://github.com/Ardupilot/mavlink spec. not the https://github.com/mavlink/mavlink spec

mark-nick-o, 2022-01-21:
sorry i fail to see what the difference is mate, im trying to send this command mavlink_send_camera_information and this is required it seems to get mixed up in the sender and no matter what i put in there the message is appearing to try to reference an index out of bounds 

how do those lengths correspond to the data array_length is it set for both char and int but length is 1 for char strings ?
orders are the order it appears in the function and is then changed here

and how do i edit this and use these functions as importing a custom library ?

i need these functons and i will attempt to fix them if you havent time.

peterbarker, 2022-01-24:
Sender:

```
#!/usr/bin/env python

import os

from pymavlink import mavutil

the_connection = mavutil.mavlink_connection(
    "tcp:localhost:5762",
    source_system=3,
    source_component=4,
)

arr_vendor = [0] * 32
arr_vendor[0] = ord("A")

arr_model = [0] * 32
arr_model[0] = ord("C")

print("Sending")
the_connection.mav.camera_information_send(
    100,
    arr_vendor,
    arr_model,
    0,
    0,
    0,
    0,
    0,
    0,
    0,
    0,
    0,
    "fred")

print("Sent")
```

receiver sees:

```
status CAMERA_INFORMATION --verbose
STABILIZE> 2022-01-24 11:55:47.39: CAMERA_INFORMATION (id=259) (link=None) (signed=False) (seq=0) (src=3/4)
    time_boot_ms: 100ms
    vendor_name: [65, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    model_name: [67, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    firmware_version: 0
    focal_length: 0.0mm
    sensor_size_h: 0.0mm
    sensor_size_v: 0.0mm
    resolution_h: 0pix
    resolution_v: 0pix
    lens_id: 0
    flags: 0
      ! CAMERA_CAP_FLAGS_CAPTURE_VIDEO
      ! CAMERA_CAP_FLAGS_CAPTURE_IMAGE
      ! CAMERA_CAP_FLAGS_HAS_MODES
      ! CAMERA_CAP_FLAGS_CAN_CAPTURE_IMAGE_IN_VIDEO_MODE
      ! CAMERA_CAP_FLAGS_CAN_CAPTURE_VIDEO_IN_IMAGE_MODE
      ! CAMERA_CAP_FLAGS_HAS_IMAGE_SURVEY_MODE
      ! CAMERA_CAP_FLAGS_HAS_BASIC_ZOOM
      ! CAMERA_CAP_FLAGS_HAS_BASIC_FOCUS
      ! CAMERA_CAP_FLAGS_HAS_VIDEO_STREAM
      ! CAMERA_CAP_FLAGS_HAS_TRACKING_POINT
      ! CAMERA_CAP_FLAGS_HAS_TRACKING_RECTANGLE
      ! CAMERA_CAP_FLAGS_HAS_TRACKING_GEO_STATUS
    cam_definition_version: 0
    cam_definition_uri: fred
```

Don't expect the array lengths in the generated code to be in the same order as those in the XML file.  Fields are re-ordered for wire-packing purposes.

mark-nick-o, 2022-01-24:
Thanks very much mate

yes it works in py2 (please note this isnt so easy to get the pip to install pymavlink but this worked)
wget https://bootstrap.pypa.io/pip/2.7/get-pip.py
python2 get-pip.py

this is what pi3 looks like you have to know what steven told me earlier to use encode on the string

            the_connection.mav.camera_information_send(
            100,
            arr_vendor,
            arr_model,
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            0,
            "fred".encode('ascii'))

peterbarker, 2022-01-25:
Thanks @mark-nick-o - I guess we can close this now?

We really ought to document this little... uh... "idiosyncracy" somewhere.  Perhaps as an example in the examples folder would make sense?  If you've got a patched version which works for both v2 and v3 that would be a handy addition to the examples folder.  Sadly we still support Py2, so we have "is_py2" code floating around to switch between using bytes() and whatnot.

mark-nick-o, 2022-02-25:
yes thanks very much you can close it, works well mate, and yes ill make the version work with python2 as well its most curl stuff and prints that will need changing, will do that as soon as i can mate

closed 2022-03-24 by peterbarker

thealienthing, 2022-08-23:
I wish there was a method for generating these arrays. I'm currently using the following method.
```
def string_to_uint8_array(string, string_length):
    string = string.ljust(string_length, '0')
    string = list(map(lambda char: ord(char), list(string)))
    return [0 if num == 48 else num for num in string]

string = "hello"
arr = string_to_uint8_array(string, 10)
print(arr)
[104, 101, 108, 108, 111, 0, 0, 0, 0, 0]

```

Is there not a built in pymavlink method or python method for converting a string to a fixed sized uint8_t array? Hate to be reinventing the wheel with some hacky method.