ArduPilot/pymavlink #621: video_stream_information send

mark-nick-o, 2022-01-13:
# video stream
    framerate = 30.0                                                          # [Hz] Frame rate.
    bitrate = 3000                                                            # [bits/s] Bit rate.
    Vflags = 3                                                                 # Bitmap of stream status flags.
    Vresolution_h = 300                                                        # [pix] Horizontal resolution.
    Vresolution_v = 400                                                        # [pix] Vertical resolution.
    rotation = 90                                                             # [deg] Video image rotation clockwise.
    hfov = 45                                                                 # [deg] Horizontal Field of view.
    stream_id = 2                                                             # Video Stream ID (1 for first, 2 for second, etc.)
    count = 4                                                                 # Number of streams available.
    stream_type = mavutil.mavlink.VIDEO_STREAM_TYPE_MPEG_TS_H264              # Type of stream.
    videoname = "vid_001"
    video_uri = "http://10.0.0.56/vids/001.mov"

    def mavlink_send_video_stream_information(self, the_connection):
        #if self.mavlink10():
        print("    !!! sending the video stream information   !!! \n")
        try:             
            the_connection.mav.video_stream_information_send(
                self.stream_id, 
                self.count, 
                self.stream_type, 
                self.Vflags, 
                self.framerate, 
                self.Vresolution_h, 
                self.Vresolution_v, 
                self.bitrate, 
                self.rotation, 
                self.hfov, 
                self.videoname, 
                self.video_uri) 
            ret = True
        except Exception as err_msg:
            print("Failed to send video stream information message : %s" % (err_msg))
            ret = False
        return ret

[image]

but gets the following error relating to a bytestream ???

stephendade, 2022-01-13:
That's a Python2 vs Python 3 issue. I assume you're running Python3?

Python3 strings need to be converted to bytes for MAVLink. See https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_asterix.py#L254 for an example.

See https://stackoverflow.com/questions/7585435/best-way-to-convert-string-to-bytes-in-python-3 for the background of the string bytes issue in Python3

mark-nick-o, 2022-01-14:
seems to work fine like this function returns good now 

    def mavlink_send_video_stream_information(self, the_connection):
        print("    !!! sending the video stream information   !!! \n")
        try:
            the_connection.mav.video_stream_information_send(
                self.stream_id,
                self.count,
                self.stream_type,
                self.Vflags,
                self.framerate,
                self.Vresolution_h,
                self.Vresolution_v,
                self.bitrate,
                self.rotation,
                self.hfov,
                #self.videoname,
                (self.videoname).encode('ascii'),
                (self.video_uri).encode('ascii'))
            ret = True
        except Exception as err_msg:
            print("Failed to send video stream information message : %s" % (err_msg))
            ret = False
        return ret

would you expect me to see this in mavlink inspector as i cant see it, the function is returning True now without error when i send it so it suggests all the parameters are good.

[image]

akvsachin, 2022-02-10:
@stephendade I wanna work on this issue. Please assign this to me and since I'm new to this repo please guide me through.

rmackay9, 2023-09-26:
@rayyanInqline786,

If you're using ArduPilot, it doesn't send these messages yet but it's high on the to-do list.

rayyanInqline786, 2023-09-28:
> @rayyanInqline786,
> 
> If you're using ArduPilot, it doesn't send these messages yet but it's high on the to-do list.

rmackay9, 2024-04-16:
@Ronan0912,

ArduPilot will forward messages received on the autopilot's serial ports to the GCS unless the SERIALx_OPTION bit has been set to disable forwarding

@rayyanInqline786,

I'm not really a pymavlink expert I'm afraid.

Ronan0912, 2024-04-16:
Thanks! All the messages looks forwarded to the GCS except those two. I will add debug traces to understand what is happening.

Ronan0912, 2024-04-16:
FYI ArduPilot is correctly forwarding the VIDEO_STREAM_INFORMATION. It is the Herelink remote/airunit connected to the ArduPilot that is filtering it (https://discuss.cubepilot.org/t/video-streaming-issue-in-custom-qgc-for-herelink/12858).