ArduPilot/pymavlink #625: Mission Upload on PX4 1.12

exabotixmpaul, 2022-02-01:
I have the following problem, when I want to upload a mission, I get the message 

```
CRITICAL:autopilot:WPM: IGN CLEAR CMD: Busy
CRITICAL:autopilot:WPM: IGN MISSION_COUNT: Busy
```

Firmware: PX4 1.12

```
wp = mavwp.MAVWPLoader()

wp.add(mavutil.mavlink.MAVLink_mission_item_message(0,0,0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 16, 0, 1, 0, 0, 0, 0, -34.364114, 149.166022, 30))
wp.add(mavutil.mavlink.MAVLink_mission_item_message(0,0,1, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 16, 0, 1, 0, 0, 0, 0, -34.365114, 149.167022, 30))
wp.add(mavutil.mavlink.MAVLink_mission_item_message(0,0,2, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 16, 0, 1, 0, 0, 0, 0, -34.363114, 149.162022, 30))

vehicle._master.waypoint_clear_all_send()
vehicle._master.waypoint_count_send(wp.count())
```

maybe I'm doing something wrong and thanks for the help

peterbarker, 2022-02-01:
On Tue, 1 Feb 2022, exabotixmpaul wrote:

> CRITICAL:autopilot:WPM: IGN CLEAR CMD: Busy
> CRITICAL:autopilot:WPM: IGN MISSION_COUNT: Busy

I'm not even sure what "Busy" has been mapped from - maybe 
"MAV_MISSION_DENIED"?!

> Firmware: PX4 1.12

Sorry, no experience running PX4 here.

> wp = mavwp.MAVWPLoader()
>
> wp.add(mavutil.mavlink.MAVLink_mission_item_message(0,0,0, mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, 16, 0, 1, 0, 0, 0, 0, -34.364114, 149.166022, 30)) [...]

Nothing popping out at me.  You could test the same code against ArduPilot 
for another opinion on your code; past that looking through the PX4 source 
code to see under what circumstatnces they will return "busy" when you're 
just trying to clear a mission is probably the next thing to do...

exabotixmpaul, 2022-02-02:
I have the same problem as you that there is no experience with px4, but it definitely works under arducopter. Thank you for the fast answer.