ArduPilot/pymavlink #641: Can arm and disarm with mavutil but can't override rc channels.

Nbelles, 2022-02-25:
Hello,

I'm currently using Pymavlink (specifically the mavutil functions) to try and arm and then override the rc channels but it is not fully working. I am able to arm the quadcopter and it spins up the motors to the default speed but then I try to have rc channels overridden but it doesn't do anything and I also don't get any errors saying that it didn't go through. Here is general idea of how my code works:

```
from pymavlink import mavutil as mavutil

m = mavutil.mavlink_connection(connection, baud=111100)
m.mav.command_long_send(
    m.target_system,
    m.target_component,
    mavutil.mavlink.MAV_CMD_COMPONENT_ARM_DISARM,
    0,
    1,
    0,
    0,
    0,
    0,
    0,
    0,
)
rc_channel_values = [65535 for _ in range(18)]
rc_channel_values[0:4] = [1900, 1900, 1900, 1900]
m.mav.rc_channels_override_send(
    m.target_system,  # target_system
    m.target_component,  # target_component
    *rc_channel_values,  # RC channel list, in microseconds.
)
```

I've been banging my head against the wall trying to reverse engineer all the code and documentation to fix it but I'm really not getting anywhere. I don't even know how to debug the whole system, it just either works or it doesn't. Any help would be greatly appreciated, thanks!

peterbarker, 2022-03-05:
On Fri, 25 Feb 2022, Nathaniel.Belles wrote:

> I'm currently using Pymavlink (specifically the mavutil functions) to try and arm and then override the rc channels but it is not fully working. I am able to arm the quadcopter and it spins up the [...]

You have to continuously send these - otherwise the vehicle would keep 
doing what it was doing if your script crashed.

Sorry, but you're also asking the wrong question; sending RC overrides to 
the vehicle is about the worst way to control it remotely (short of 
getting a broomstick and pushing it around, but not by much).

You really want to look at "guided/guided_nogps" (or "offboard") mode; you 
then tell the aircraft to move to a point in space or come to a specific 
attitude, that sort of thing.

Nbelles, 2022-03-05:
I took out the while loop from my simplification of the code, I understand that I'm supposed to repeatedly send the commands. 

I'm using this for a research application where the quadcopter is tethered over USB and we have a control system on a computer that is handling all the real-time high power computation bits. The data about position, velocity and accelerations are being directly sent from the quad into that real-time controller and is outputting desired RC inputs. The intended end goal is that the outputs of those RC inputs would be sent back to the quadcopter. 

I understand it's not really the best way to control a quadcopter for performance reasons but for us, it is the best way to test these new algorithms using more computationally complex methods for stable flight. That's what brought me to the sending RC overrides because it essentially gives override of the RC inputs when not in stabilized mode.

peterbarker, 2022-03-24:
> I'm using this for a research application where the quadcopter is tethered over USB and we have a control system on a computer that is handling all the real-time high power computation bits. The [...]

Sorry, but pretty sure this makes no sense.  Stable flight can't be achieved by sending "RC inputs", as those are simply telling the vehicle where to fly, not how to fly.  Now specifying your own *servo outputs* would more match your description here - and doing that via mavlink is going to be interesting to say the least.  But using RC input is going to put a lot of ArduPilot's code between you and the motor outputs.

What level are you really seeking to control the vehicle at?  Are you hoping to specify the throttle one each motor individually?  That's doable, but the latencies involved are horrendous so you will struggle to get stable flight.  You need to get the data out to your companion computer and back into the autopilot with 10s-of-milliseconds latency at worst.

You might reconsider running the algorithm on-board; H7 is really quite a powerful device.  You might also consider a Linux-based autopilot.

peterbarker, 2022-03-24:
@Nbelles if you can give me a minimal example which doesn't work for you I might be able to tell you what's wrong with your messaging.

rmackay9, 2022-03-24:
Probably not helpful but just in case we have some information here on how a companion computer can control a Copter while it is in Guided mode using mavlink https://ardupilot.org/dev/docs/copter-commands-in-guided-mode.html