ArduPilot/pymavlink #660: extracting a bunch of parameters from tlog file

emmaemmstein, 2022-04-12:
I saw this code [here](https://github.com/ArduPilot/pymavlink/blob/master/tools/mavparms.py) which extracted param_id and their values. I wanted to know how I could modify it to print out more data (like Longitude, altitude, latitude etc) and their resulting values. I know data like this exist because I can access them from their csv format but they are not printed out using the mavparms.py code. Pardon my ignorance as I am new to using pymavlink.

Thanks

emmaemmstein, 2022-04-18:
I Found my way around it

closed 2022-04-18 by emmaemmstein

brunoolivieri, 2022-04-18:
@emmaemmstein My I ask how? :-)

emmaemmstein, 2022-04-18:
I used the '.recv_match()'  method. No argument was passed in it so it
printed all the message