ArduPilot/pymavlink #688: How to use multiple types for mavlogdump.py

anshulsood-02, 2022-06-13:
I am trying to use this command: 
`mavlogdump.py --types=BAT,BARO --format=csv L1.BIN > L1.csv`

But it only generates a csv of the first type mentioned.
The code mentions: types of messages (comma separated with wildcard)
but I am not sure if it works or if I am typing it wrong.

peterbarker, 2022-06-23:
@anshulsood-02 you cn only dump from one message type when generating a CSV.

We could probably come up with a rule on when to emit a CSV line, but we don't have one ATM.  You don't want to emit for every (e.g.) `BAT` and `BARO` line, for example.  Most sensible would probably emit any time you've seen all message types and get an instance of the first message, or something along those lines.

shancock884, 2024-01-31:
In the current implementation of mavlogdump.py, you get a clear message if you try to specify multiple types when dumping to CSV:
```
$ mavlogdump.py --format=csv --types=UNIT,MULT ArduPlane-Soaring-00000045.BIN 
Need exactly one type when dumping CSV from bin file
```
I don't know if this would count as a 'fix' for this issue? - at least the restriction is made clear to the user in a nice way.

(See also: https://github.com/ArduPilot/pymavlink/issues/905#issuecomment-1915684598)