ArduPilot/pymavlink #701: Wireshark showing 0 for negative int32

hamishwillee, 2022-07-06:
With mission items with negative y (negative int32 value), the wireshark plugin incorrectly shows 0, even though the data is clearly not 0 (as seen in the hex dump)

[image]

This is cross posted from https://github.com/mavlink/mavlink/issues/1862

peterbarker, 2022-07-06:
@PurkkaKoodari any ideas on this one?  I figure your recent work on this generator makes you the most likely to nail this in short order :-)

PurkkaKoodari, 2022-07-07:
I haven't been working on the previous ArduPilot-based project in over a year, so I my experience isn't exactly fresh, but I'll take a look.

@hamishwillee Can you share a `.pcapng` with this packet? The code I have installed locally - not sure if it matches upstream - seems to be parsing negative int32_t's just fine. (You can use File->Export Specified Packets if you want to limit the amount of data shared.)

[image]

hamishwillee, 2022-07-07:
@PurkkaKoodari Thanks! I'm actually just cross posting this on behalf of @ThomasDebrunner - Thomas, could you please provide a .pcappng with the packet to allow easier testing?

ThomasDebrunner, 2022-07-08:
Just tried to reproduce again and get a pcapng, but it seems to be working now on my Ubuntu. I spottet the problem on my mac, so maybe it is platform related? Will try again on the mac later.

PR #944 (2024-05-01): Mavgen WLUA: Fix incorrect display of negative ints on mac
This PR is to fix https://github.com/mavlink/mavlink/issues/1862, fix https://github.com/mavlink/mavlink/issues/2073 and fix #701. (In particular https://github.com/mavlink/mavlink/issues/2073 includes the discussion about how it was figured out!).

There is a bug in wireshark, whereby a double is cast to an unsigned int when it should be a signed int, before being used as a signed int. On Linux this works, but on Mac negative values get cut to zero ([here is why](https://embeddeduse.com/2013/08/25/casting-a-negative-float-to-an-unsign [... cut]
closed 2024-05-01 by peterbarker