ArduPilot/pymavlink #714: pymavlink not working with asyncio

patrickelectric, 2022-08-02:
Something simple as having multiple connection between corroutines appears to not work with pymavlink.

```python
import asyncio
from pymavlink import mavutil

PORT = 6666
CONNECTION_STRING = f'udpout:127.0.0.1:{PORT}'

def get_mavlink_connection(system_id: int, component_id: int):
    return mavutil.mavlink_connection(
        CONNECTION_STRING,
        source_system=system_id,
        source_component=component_id,
        autoreconnect=True,
    )

async def heartbeat_loop(system_id: int, component_id: int):
    mavlink_connection = get_mavlink_connection(system_id, component_id)
    while True:
        await asyncio.sleep(1)
        print(f"{system_id}/{component_id}: sending heartbeat")
        mavlink_connection.mav.heartbeat_send(
            0,
            mavutil.mavlink.MAV_TYPE_CAMERA,
            mavutil.mavlink.MAV_AUTOPILOT_GENERIC,
            0,
            mavutil.mavlink.MAV_STATE_STANDBY,
            3
        )

async def receive_loop(system_id: int, component_id: int):
    mavlink_connection = get_mavlink_connection(system_id, component_id)
    while True:
        await asyncio.sleep(1)
        print("Receive..")
        incoming = message = mavlink_connection.recv_match()
        if incoming is None:
            continue
        message = incoming.to_dict()
        print(f"{system_id}/{component_id}: got message: {message['mavpackettype']}")

def create_component_tasks(loop: asyncio.AbstractEventLoop, system_id: int, component_id: int) -> [asyncio.Task]:
    return [
        loop.create_task(heartbeat_loop(system_id, component_id)),
        loop.create_task(receive_loop(system_id, component_id)),
    ]

if __name__ == '__main__':
    loop = asyncio.new_event_loop()

    loop.run_until_complete(asyncio.wait(
        create_component_tasks(loop, mavutil.mavlink.MAV_COMP_ID_AUTOPILOT1, mavutil.mavlink.MAV_COMP_ID_CAMERA) +
        create_component_tasks(loop, mavutil.mavlink.MAV_COMP_ID_AUTOPILOT1, mavutil.mavlink.MAV_COMP_ID_CAMERA2)
    ))
    loop.close()
```

patrickelectric, 2022-08-02:
Any tip @khancyr @peterbarker ?

khancyr, 2022-08-02:
Should probably await on a simple recv(). Recv_match is a bit complicated.

I think a guy share an async lib for pymavlink on discuss, I lack of knownledge with async so you may find it usefull

patrickelectric, 2022-08-03:
I got it working in the end, but it was necessary to create the connections outside the async functions, from my tests here, it appears that creating such connections results in some crazy problems.
https://gist.github.com/patrickelectric/d6264fe309d01b44d5540d75cc45cb83

closed 2022-08-03 by patrickelectric