ArduPilot/pymavlink #716: fromstring is removed in python 3.9

mnumanuyar, 2022-08-10:
fromstring is removed from python 3.9. ([changelog](https://docs.python.org/3/whatsnew/3.9.html))
related issue: https://github.com/python/cpython/issues/83097

currently mavutil has an instance of it
https://github.com/ArduPilot/pymavlink/blob/ebc4bd46f9a0a36ecbe73aec48d08e36e0e1a3d0/mavutil.py#L2286

running it produces error: 
"'array.array' object has no attribute 'fromstring' python"

I changed the line as 
```python
            if type(buf) == bytes: 
                byte_buf.frombytes(buf)
            elif type(buf) == str: 
                byte_buf.fromlist(list(buf.encode('utf8')))
```

in my local, which solved error. But I have no clue if this is correct, and I have not tested it well

alehed, 2022-11-02:
This error should have been resolved by #666 which is now merged.

mnumanuyar, 2022-11-03:
that line is still in the master branch, did the merge somehow made that line unaccesible ?

alehed, 2022-11-03:
Ah, true. I was thinking about the generated code.

closed 2022-11-09 by tridge