ArduPilot/pymavlink #720: Pymavlink version 2.4.33 doesn't allow for mode changes

bayernfan31, 2022-08-16:
I am using mavproxy to communicate with dronekit-sitl on a windows machine. I am not able to change from Stabliize to other modes in my script, or directly in the mavproxy command line. Other forums discussed switching to pymavlink 2.4.8 but that is not an option here. Any other ideas for how I can get it to switch modes?

amilcarlucas, 2025-12-30:
You can only switch to modes that require GPS position if you have GPS position available.

So to solve this, enable fence in ardupilot. That will only allow takeoff when you have GPS position, and if you have GPS position, you will be able to switch modes.

peterbarker, 2025-12-30:
On Tue, 30 Dec 2025, Amilcar Lucas wrote:

> So to solve this, enable fence in ardupilot. That will only allow takeoff when you have GPS position, and if you have GPS position, you will be able to switch modes.

More recent versions of ArduPilot include ARM_NEED_LOC:

     // @Param: NEED_LOC
     // @DisplayName: Require vehicle location
     // @Description: Require that the vehicle have an absolute position 
before it arms.  This can help ensure that the vehicle can Return To 
Launch.
     // @User: Advanced
     // @Values{Copter,Rover}: 0:Do not require location,1:Require Location