ArduPilot/pymavlink #730: TypeError on .bin file to json when running mavlogdump.py

nedraebb, 2022-09-14:
The command I am using is **python mavlogdump.py --format json filename.bin > filename.bin.json**

Raises TypeError(f'Object of type {o.__class__.name__}')
TypeError: Object of type bytes is not JSON serializable

I have tried this on multiple .bin files and it prints about 1400 hundred lines into the output file before it fails.

peterbarker, 2022-09-16:
@nedraebb could you test the linked PR, please?  Note the caveat that file content won't be recoverable afterwards.

nedraebb, 2022-09-16:
@peterbarker Tested this out on a few different .bin files and it is working. Thanks!

peterbarker, 2022-09-19:
> @peterbarker Tested this #731 on a few different .bin files and it is working. Thanks! I am confused as to what you mean by the file content won't be recoverable after.

I'm not attempting to preserve the file content.  To represent it in JSON would require base64-encoding it or similar, in case you wanted to extract the file content from the JSON.  Can't do binary in JSON files, must be valid utf-8 AFAIK.

peterbarker, 2022-09-19:
... and to be clear, ArduPilot embeds some file content within the dataflash logs.  It's in the `FILE` dataflash log message.

PR #731 (2022-09-22): mavlogdump.py: don't die when handling bytestrings in dataflash logs
json doesn't permit non-utf-8, but our logs contain actual file contents in some fields.  This isn't going to work out well.

Past this the file content will not be recoverable; we could, perhaps, encode into base64 or similar instead of doing the conversion.

Closes #730
closed 2022-09-22 by peterbarker