ArduPilot/pymavlink #742: TCP Cannot cannot check for tcp disconnection to recover

thealienthing, 2022-10-21:
Working on a project that uses mavlink router with tcp connections. Occasionally mavlink-router must be closed without warning and sometimes closes because of some error. My software that connects via tcp needs to be able to recover from these disconnections. There doesn't seem to be a way to detect that an instance of the **mavtcp** class has had it's connection closed unless auto_reconnect has been set to True.

Here are the methods concerning disconnection:
```
    def close(self):
        self.port.close()

    def handle_disconnect(self):
        print("Connection reset or closed by peer on TCP socket")
        self.reconnect()

    def handle_eof(self):
        # EOF
        print("EOF on TCP socket")
        self.reconnect()

    def reconnect(self):
        if self.autoreconnect: # <--- Unless autoreconnect is enabled, the class member port is never explicitely called so you can't check if it is closed
            print("Attempting reconnect")
            if self.port is not None:
                self.port.close()
                self.port = None # <--- Only after reconnect attempts have run out will the port be set to None
            self.do_connect()
```

If you have autoreconnect enabled, you **can** check that the port is closed by checking if the port is None or by catching a ConnectionRefusedError. But when autreconnect is not set, no exceptions can be caught and port is not set to None. Instead you get a massive stdout print out of "EOF on TCP socket"

Would the pymavlink project maintainers be interested in a fix for this. I'll probably fix this on my own fork of pymavlink. Let me know.

thealienthing, 2022-10-24:
If no responds to this with a veto or reason this shouldn't be fixed, I'll put in a pull request with an example in a couple days :)

fockboi-lgtm, 2025-06-18:
@thealienthing what was the fix you came up for this issue?