ArduPilot/pymavlink #754: Very slow data transsmission via Radio Modem

szymek1, 2022-11-24:
Hello,
I'm working on application which receives drone's telemetry through radio signals (with RFD 868x radio modem) and then sends them via UDP to another application. My code for receiving telementry and sending it is written in Python. When my connection is wired I can set in `mavutil.mavlink_connection('com7', baud=57600)` or leave it default baud=115200 and in both cases I have very good speed. Btw. I'm sending only 6 floats so not much data...
Unfortunately, when my connection is wireless, so this time with radio modem, I have to set baud to 57600 and the speed rate is terrible. What I find interesting is that when data is received wireless but with MissionPlanner software I don't record such a terrible speed as when the software is pure Python.
My question is: can it be that much of a difference between MissionPlanner utilization of mavlink and pymavlink? Is there any place I can optimize the speed of transmission apart from:
`mavutil.mavlink_connection('com7', baud=57600)
`
and
`mav.request_data_stream_send(self.droneConnection.target_system, 
            self.droneConnection.target_component, mavutil.mavlink.MAV_DATA_STREAM_ALL, 200, 1)`
Simply increasing the rate from 200 to for example 1000 in `mav.request_data_stream_send` doesn't help.

peterbarker, 2022-11-28:
On Thu, 24 Nov 2022, Szymon Bogus wrote:

> My question is: can it be that much of a difference between MissionPlanner utilization of mavlink and pymavlink? Is there any place I can optimize the speed of transmission apart from:
> mavutil.mavlink_connection('com7', baud=57600)

Use hardware flow control.  Ensure MissionPlanner is either not setting 
stream rates or reduce the streamrates it is setting.

szymek1, 2022-11-30:
I fixed that with:

           def Request_message_interval(self, message_id: int, frequency_hz: float):
            
              self.droneConnection.mav.command_long_send(
                  self.droneConnection.target_system, self.droneConnection.target_component,
                  mavutil.mavlink.MAV_CMD_SET_MESSAGE_INTERVAL, 0,
                  message_id, 
                  1e6 / frequency_hz,
                  0, 0, 0, 0, 
                  0, 
              )

closed 2022-11-30 by szymek1