ArduPilot/pymavlink #768: Px4 Pymavlink, Takeoff not happening

Murali-IsPagro, 2022-12-29:
Hello,

I tried arming and takeoff using pymavlink from my Ubuntu 18.04 system.I could able to Arm the vehicle But , Drone is not taking off, propellers are just spinning. Am I missing something. I have attached the log file. I have opened mavproxy.py in one terminal in second terminal i have run the code. Any help will be appreciated.

PX4 v1.13.0
Ubuntu 18.04 LTS

Code:
  
    #code

    from pymavlink import mavutil   
    connection_string = '127.0.0.1:14540'
    the_connection = mavutil.mavlink_connection('udp:'+connection_string)
    data_rate = 1 
    the_connection.mav.request_data_stream_send(the_connection.target_system, the_connection.target_component,
                                            mavutil.mavlink.MAV_DATA_STREAM_ALL, data_rate, 1)

    the_connection.wait_heartbeat()
    print("Heartbeat from system (system %u component %u)" %
          (the_connection.target_system, the_connection.target_component))

    def arm(the_connection):
	    the_connection.mav.command_long_send(the_connection.target_system, the_connection.target_component,
        	                                 mavutil.mavlink.MAV_CMD_COMPONENT_ARM_DISARM, 0, 1, 0, 0, 0, 0, 0, 0)

	msg = the_connection.recv_match(type='COMMAND_ACK', blocking=True)
	print(msg)

    def takeoff(the_connection):
	    the_connection.mav.command_long_send(the_connection.target_system,                                                                                                                         the_connection.target_component,mavutil.mavlink.MAV_CMD_NAV_TAKEOFF, 0, 0, 0, 0, 0, 0, 0, 100)

	    msg = the_connection.recv_match(type='COMMAND_ACK', blocking=True)
	    print(msg)

    arm(the_connection)

    takeoff(the_connection)

Log file:
[Click here for log](https://review.px4.io/plot_app?log=126790bb-a86e-4530-adba-f93e33263b34)

Czw007, 2023-11-22:
Hi, I also encountered the same problem, have you solved this problem?

peterbarker, 2023-12-14:
So there's one bug in script; the first `recv_match` is mis-indented so runs before the command is issued!

After correcting that I ran the script against ArduPilot (in guided mode) and it worked just fine.

Closing this as it's not a problem with pymavlink :-)

closed 2023-12-14 by peterbarker