ArduPilot/pymavlink #771: How does the code ignore the extensions when computing the CRC?

JonasVautherin, 2023-01-26:
### Summary

The [MAVLink docs](https://mavlink.io/en/guide/define_xml_element.html#message_extensions) say that _"MAVLink 2 defines extension fields, which can be added to an existing message without breaking binary compatibility for receivers that have not been updated"_.

I am trying to understand how that works.

### CRC_EXTRA

I understand that the CRC_EXTRA (which checks the message _definition_) does not check the extension fields (see [here](https://github.com/ArduPilot/pymavlink/blob/master/generator/mavparse.py#L458)). This means that when using extensions, one could change the message definition in such a way that the receiver reads garbage without realizing it. Fine, I can live with that.

### CRC

I was told (thanks @hamishwillee :hugs:) that the extensions are also ignored in the CRC. Meaning this time that they could be corrupted before reaching the receiver, and the receiver would not realize it. Again, let's live with that. My problem is that I don't find in the code how that is done.

#### Java

Looking at the Java code, I [see here](https://github.com/ArduPilot/pymavlink/blob/master/generator/java/lib/Parser.java#L172-L174) that the payload is read until the `length` byte:

```java
                if (m.payloadIsFilled()) {
                    state = MAV_states.MAVLINK_PARSE_STATE_GOT_PAYLOAD;
                }
```

where `m.payloadIsFilled()` is:

```java
    public boolean payloadIsFilled() {
        return payload.size() >= len;
    }
```

and `len` is [set by the length byte in the message](https://github.com/ArduPilot/pymavlink/blob/master/generator/java/lib/Parser.java#L99) (i.e. that includes the extensions if the sender had extensions).

And of course it [computes the CRC](https://github.com/ArduPilot/pymavlink/blob/6b1e0c78d3cd6a53ea42d64e4a0f9d22aae1adf3/generator/java/lib/Messages/MAVLinkPayload.java#L178) based on that length:

```java
boolean crcGen = m.generateCRC(m.payload.size());
```

This is for parsing, but that's similar for encoding (though I can't find code to link to online). It essentially does:

```java
        for (int i = 0; i < payloadSize; i++) {
            crc.update_checksum(payload.getByte());
        }
```

with `payloadSize` being:

```java
        if (isMavlink2) {
            payloadSize = mavTrimPayload(payload.payload.array());
            bufLen = MAVLINK2_HEADER_LEN + payloadSize + 2;
        } else {
            payloadSize = payload.size();
            bufLen = MAVLINK1_HEADER_LEN + payloadSize + 2;
        }
```

Question here: where is the code ignoring the extensions when computing the CRC?

#### C++

It is similar in C++, except that [here](https://github.com/mavlink/c_library_v2/blob/master/common/mavlink_msg_manual_control.h#L20-L21) I see two interesting values:

```cpp
#define MAVLINK_MSG_ID_MANUAL_CONTROL_LEN 18
#define MAVLINK_MSG_ID_MANUAL_CONTROL_MIN_LEN 11
```

where, according to the [MANUAL_CONTROL](https://mavlink.io/en/messages/common.html#MANUAL_CONTROL) definition, 18 is the length of the payload with extensions, and 11 without them. MAVLink v1 does not support extensions, so it uses `MIN_LEN` (see [here](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L235)). But v2 uses `LEN`, which includes the extensions:

```cpp
msg->len = mavlink1?min_length:_mav_trim_payload(_MAV_PAYLOAD(msg), length);
```

That's all fine, but then again, it [computes the CRC](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L266) check on the length `msg->len`:

```cpp
crc_accumulate_buffer(&checksum, _MAV_PAYLOAD(msg), msg->len);
```

I would expect `min_length` here :thinking:. I don't really get what [`_MAV_PAYLOAD`](https://github.com/mavlink/c_library_v2/blob/master/mavlink_types.h#L157) does, though (it returns a pointer to the beginning of the payload maybe? :man_shrugging:). Anyway it is used to define `msg->len`, so I guess it should contain the extensions.

This is when encoding, but again, that's similar when parsing: it seems to [compute the CRC on the whole payload](https://github.com/mavlink/c_library_v2/blob/master/mavlink_helpers.h#L727-L734):

```cpp
	case MAVLINK_PARSE_STATE_GOT_MSGID3:
		_MAV_PAYLOAD_NON_CONST(rxmsg)[status->packet_idx++] = (char)c;
		mavlink_update_checksum(rxmsg, c);
		if (status->packet_idx == rxmsg->len)
		{
			status->parse_state = MAVLINK_PARSE_STATE_GOT_PAYLOAD;
		}
		break;
```

So I end up on the same question here: how does the C++ implementation ignore the extensions in the CRC check?

Sorry for the long question :see_no_evil:, and thanks in advance for your insights!

hamishwillee, 2023-01-26:
@peterbarker @tridge The TLDR of this is that the CRC calculation appears to be run over the whole payload, including both original fields and extension fields. This will be a problem for an updated sender and a non-updated recipient because non-zero values for those extension fields will result in a different CRC than a non-updated recipient (which will not see those fields at all).
It should not be a problem for other cases because the sender and receiver with otherwise see the same fields values.

The CRC_EXTRA is calculated only on the extension fields for this reason.

This appears to affect C and Java libraries. Possibly others. Hope we're missing something.

julianoes, 2023-01-26:
> I was told (thanks @hamishwillee hugs) that the extensions are also ignored in the CRC.

I was assuming the CRC runs over the whole payload. It's just a CRC to tell you that the bytes have been transmitted correctly or not.

The CRC_EXTRA would then be to make sure that you have the same message definition on both sides, right?

JonasVautherin, 2023-01-26:
> I was assuming the CRC runs over the whole payload.

Oooooh that actually makes sense :man_facepalming:. Even if it does not understand the extensions, the receiver knows how to compute the CRC... only after the message has been CRC-validated, the receiver will start using it, and there it will ignore the extension fields that it doesn't know.

Well at least I got familiar with the codebase :grin:

hamishwillee, 2023-01-26:
@julianoes The thing is, the recipient accumulates the received bytes to get the CRC, then accumulates the CRC_EXTRA. The result is a final CRC that you compare to what is in the message, and that tells you the message is both not corrupt and matching the original message definition. 

AFAIK there is no way to test that the CRC matches but the CRC_EXTRA does not (or visa versa). So in this case where the CRC might not match, you can still have a problem.

hamishwillee, 2023-01-26:
Ignore me, I know not what I say. Both ends run the CRC on the same buffer. I wasn't getting that the recipient doesn't know what to do with the extra fields, but it is still getting them. Please close.