ArduPilot/pymavlink #780: Cannot move servo on pichawk cube orange with pymavlink

sezer-muhammed, 2023-02-17:
Hi, I cannot move servo or send data to my pixhawk.
I can control my pixhawk via RC. I tried to set pixhawk's mode to "Guided, Stabilize, auto, manuel, FBWA" non of them worked.

there is my code:

```
from pymavlink import mavutil
from pymavlink.dialects.v20 import common as mav_msgs

import os

os.environ["MAVLINK20"] = "1"
mavutil.set_dialect("common")

# Start a connection listening on a UDP port
pixhawk = mavutil.mavlink_connection("COM6", baud=115200, dialect="common")

pixhawk.wait_heartbeat()
print("Heartbeat from system (system %u component %u)" % (pixhawk.target_system, pixhawk.target_component))

while True:
    pixhawk_msg = pixhawk.recv_match()
    if isinstance(pixhawk_msg, mav_msgs.MAVLink_servo_output_raw_message):
        print(pixhawk_msg)
        pixhawk.mav.command_long_send(
        pixhawk.target_system, pixhawk.target_component,
        mavutil.mavlink.MAV_CMD_DO_SET_SERVO,
        0,            # first transmission of this command
        10,  # servo instance, offset by 8 MAIN outputs
        1200, # PWM pulse-width
        0,0,0,0,0     # unused parameters
    )
```

And the output:

```
SERVO_OUTPUT_RAW {time_usec : 279529543, port : 0, servo1_raw : 1000, servo2_raw : 1493, servo3_raw : 1280, servo4_raw : 1284, servo5_raw : 1493, servo6_raw : 1000, servo7_raw : 0, servo8_raw : 0, servo9_raw : 0, servo10_raw : 0, servo11_raw : 0, servo12_raw : 0, servo13_raw : 0, servo14_raw : 0, servo15_raw : 0, servo16_raw : 0}
SERVO_OUTPUT_RAW {time_usec : 280029503, port : 0, servo1_raw : 1000, servo2_raw : 1493, servo3_raw : 1280, servo4_raw : 1284, servo5_raw : 1493, servo6_raw : 1000, servo7_raw : 0, servo8_raw : 0, servo9_raw : 0, servo10_raw : 0, servo11_raw : 0, servo12_raw : 0, servo13_raw : 0, servo14_raw : 0, servo15_raw : 0, servo16_raw : 0}
SERVO_OUTPUT_RAW {time_usec : 280549596, port : 0, servo1_raw : 1000, servo2_raw : 1493, servo3_raw : 1280, servo4_raw : 1284, servo5_raw : 1493, servo6_raw : 1000, servo7_raw : 0, servo8_raw : 0, servo9_raw : 0, servo10_raw : 0, servo11_raw : 0, servo12_raw : 0, servo13_raw : 0, servo14_raw : 0, servo15_raw : 0, servo16_raw : 0}
SERVO_OUTPUT_RAW {time_usec : 281029519, port : 0, servo1_raw : 1000, servo2_raw : 1493, servo3_raw : 1280, servo4_raw : 1284, servo5_raw : 1493, servo6_raw : 1000, servo7_raw : 0, servo8_raw : 0, servo9_raw : 0, servo10_raw : 0, servo11_raw : 0, servo12_raw : 0, servo13_raw : 0, servo14_raw : 0, servo15_raw : 0, servo16_raw : 0}
```

amilcarlucas, 2024-01-22:
Are you using ArduPilot 4..4.4? You should.

rmackay9, 2024-01-23:
It might be good to test using MAVProxy and the commands on this wiki page https://ardupilot.org/dev/docs/mavlink-move-servo.html