ArduPilot/pymavlink #792: Inconsisten arm/disarm state

Cryoschrome, 2023-04-05:
Hi, 
I am trying to read the arm/disarm state of my drone, running ardupilot, using using pymavlink connected to the drone via UDP.
A [herelink](https://docs.cubepilot.org/user-guides/herelink/herelink-overview) controller is also connected to the drone. With there herelink connected, the pymavlink reports inconsistent arm/disarm state. It constantly fluctuates between arm and disarm. 
[image]

QGC and Mission Planner does not have this inconsistency. Disconnecting the herelink eliminated the fluctuation.

On testing with MAVProxy with herelink shows addition vehicles.
[image]

Is there a way to ignore the addition vehicle detected?

peterbarker, 2023-04-13:
On Wed, 5 Apr 2023, Cryoschrome wrote:

> Is there a way to ignore the addition vehicle detected?

https://github.com/ArduPilot/pymavlink/pull/794 may fix the problem.

SourabhPrasad, 2023-04-27:
Hey, that fixed the issue. Thanks!

closed 2023-04-27 by SourabhPrasad