ArduPilot/pymavlink #815: Wireshark Wlua does not dissect flags

hamishwillee, 2023-05-10:
An error is shown if you try to inspect a message, such as HEARTBEAT, that has flags, and you can't inspect any fields after that flag. For example in HEARTBEAT this is the mode flag:

[image]

There are two issues. Firstly, the `value` returned by `tree:add_le` is `nil`, so what gets passed into `dissect_flags_MAV_MODE_FLAG` cannot be dissected. The docs say `tree:add_le` should return a value. My assumption is that this is failing for the reason given after the code. Note that this is the only place the value is actually used, which is why we don't see the error elsewhere.

```
    field_offset = offset + 6
    subtree, value = tree:add_le(f.HEARTBEAT_base_mode, padded(field_offset, 1))
    dissect_flags_MAV_MODE_FLAG(subtree, "HEARTBEAT_base_mode", value)
```

I "fixed" this error by passing in `padded(field_offset, 1)` to `dissect_flags_MAV_MODE_FLAG` instead of `value`. Wireshark now fails with the error that the data passed in is "userData" - i.e. an arbitrary C data that it can't interpret. That's correct, and probably why `nil` got returned from the `tree:add_le`.

The solution would be to cast the data to a number before passing it to `tree:add_le`. HOwever I don't know how to do this elegantly.

This is cross posted (and summarised) from https://github.com/mavlink/mavlink/issues/1978

hamishwillee, 2023-05-10:
FYI @auturgy can you seek expertise in the dev call. @peterbarker already pointed me to @rotu

rotu, 2023-05-11:
I think I fixed this in https://github.com/rotu/pymavlink/tree/lua-bitflags but GitHub is being annoying and not letting me open a PR. Can you give this a shot?

hamishwillee, 2023-05-11:
I certainly will. Maybe not until the weekend. Thank you very much

rotu, 2023-05-15:
I have some tentative test code but I need a pcap to add to the testing script. got data?

hamishwillee, 2023-05-15:
I don't know sorry. I planned to create your PR, test it, and then see what else was required.

PR #817 (2023-09-05): Fix Wireshark dissector and add snapshot tests
Handle bit flags in Wireshark dissector (fix #815, fix #726)
Add tests for Wireshark (via `tshark` binary and `pytest` via the `syrupy` snapshot library)
Remove defunct mav2pcap.py script
closed 2023-09-05 by peterbarker