ArduPilot/pymavlink #825: The generated java code doesn't build

pvalsecc, 2023-06-14:
If I try to generate the latest definition XMLs in Java, the generated code doesn't build. Some enum values are bigger than the max int value and the compiler complains about that.

For example, in common.xml, we have MAV_SYS_STATUS_EXTENSION_USED=2147483648

As a work around, I've patched this project like that:
```patch
diff --git a/pymavlink/generator/mavgen_java.py b/pymavlink/generator/mavgen_java.py
index 8e5086a6..87e599bc 100644
--- a/generator/mavgen_java.py
+++ b/generator/mavgen_java.py
@@ -34,7 +34,7 @@ package com.MAVLink.enums;
  * ${description}
  */
 public class ${name} {
-${{entry:   public static final int ${name} = ${value}; /* ${description} |${{param:${description}| }} */
+${{entry:   public static final int ${name} = (int)${value}L; /* ${description} |${{param:${description}| }} */
 }}
 }
             ''', en)

```

Ugly, but does the job...

Zhumos, 2023-07-31:
So I just had a look at the history of this file and it looks like there might have been a regression.

#639 changed the enum underlying type from int to long (commit: https://github.com/ArduPilot/pymavlink/pull/639/commits/53f0a82277bc702de3e771d984ea4456b72e646e) by @TSC21 

#694 reverted the type back to an int (commit: https://github.com/ArduPilot/pymavlink/pull/694/commits/bb30725b86c9f1b54b640f9fe9ee34288e64eb7c) by @rrr6399 

Perhaps @rrr6399 can comment whether the change was intentional or not.

shancock884, 2023-12-17:
There appear to be two near-identical Open PRs that would offer a fix for this: #744 & #678.
They both propose to use 'long' data type only for any enum with a value greater than 2147483647, and using int for those below (this being the max positive value of a signed 32 bit int).

I think the approach in these PRs is safer than using: "(int)${value}L", because:
 - "int ${name} = (int)2147483648L" effectively puts -2147483648 into the integer - which is generally OK as it has the right bit pattern.
 - but if this integer is ever assigned to a 'long', the high 32 bits would be filled with 1s and not 0s to maintain the number as negative.
Note that there are two 64-bit bitfields in the current XML files and while neither of these currently use bits 31 or above, it is perfectly possible that they may at some point.

peterbarker, 2024-01-10:
@pvalsecc are you happy for this issue to be closed?

I might close it now - if that wasn't the correct thing to do, please reopen.

Thanks @shancock884

closed 2024-01-10 by peterbarker