ArduPilot/pymavlink #849: Invalid syntax

anunarayan13, 2023-08-11:
from pymavlink.dialects.v10 import ardupilotmega as MAV_APM
  File "/usr/local/lib/python2.7/dist-packages/pymavlink-2.4.39-py2.7.egg/pymavlink/dialects/v10/ardupilotmega.py", line 56
    def __init__(self, buf: Optional[Sequence[int]] = None) -> None:
                                    ^
SyntaxError: invalid syntax

Please resolve

shancock884, 2023-12-17:
From your path it looks like you are using python2.7 - the website suggests that pymavlink is only supported from Python 3.5 onwards, so maybe this is the reason. (to be confirmed by someone more authoritative than me...)

Menny11, 2023-12-19:
pymavlink is used by dronekit and there is only 2.7 python supported(( I don't now why the support for py2 in the pymavlink is removed(

khancyr, 2023-12-19:
py2 is deprecatead since some years already and it become too hard to maintain compatiblity with it as most project are dropping py2, so ArduPilot move to python3 only and update most of its tools for this.

furkanisikay, 2024-01-28:
please run this command on your terminal to using pymavlink 2.4.37. it going to fix your problem.

pip install --force-reinstall -v "pymavlink==2.4.37"

peterbarker, 2024-10-27:
Closing this as no response from @anunarayan13

peterbarker, 2024-10-27:
(thanks all!)

closed 2024-10-27 by peterbarker