ArduPilot/pymavlink #852: Wireshark Wlua dissection of 64-bit flag fields

shancock884, 2023-09-05:
I used the newly merged Wireshark dissection update (pr #817 - great feature!), and noticed an LUA error on the AUTOPILOT_VERSION message.
I believe it is because this particular flag field is a uint64, rather than a uint32 (which most other flag fields are).

[image]

A bit of grepping shows that there appear to be only 2 fields in the whole ardupilotmega scope to which this issue may apply:
```
grep -A2 "value = tvbrange:le_uint64" mavlink_2_ardupilotmega.lua | grep -B2 "dissect"
    value = tvbrange:le_uint64()
    subtree = tree:add_le(f.AUTOPILOT_VERSION_capabilities, tvbrange, value)
    dissect_flags_MAV_PROTOCOL_CAPABILITY(subtree, "AUTOPILOT_VERSION_capabilities", tvbrange, value)
--
    value = tvbrange:le_uint64()
    subtree = tree:add_le(f.GENERATOR_STATUS_status, tvbrange, value)
    dissect_flags_MAV_GENERATOR_STATUS_FLAG(subtree, "GENERATOR_STATUS_status", tvbrange, value)
``` 

P.S. Sorry for the timing of this, so quickly after the merge!

hamishwillee, 2023-09-05:
Arrrrgggg. @rotu might well not want to look at this. Do you want to take a shot at it?

rotu, 2023-09-06:
Feel free to have a crack at it. A big part of my PR was making sure the code has good test infrastructure so hopefully that will make it easier to confidently change this code!

shancock884, 2023-09-10:
I've done a bit of digging around, but I can't see the 'proper' solution right now. I'll record findings here, so they aren't lost....

I think the fundamental issue is with Wireshark support for UInt64s, in a couple of areas, which would need to be raised on their developers area...

1/ As per the error shown in the first post, Wireshark does not allow a tree:add_le to be called on a Boolean ProtoField, with a UInt64 value. I think this is due to the call to wslua_tointeger() code in wslua_tree.c.

2/ Also, I wasn't able to set up a ProtoField.bool with a UInt64 bitmask (which would be necessary to use the high 32-bits). e.g.:
```
f.AUTOPILOT_VERSION_capabilities_flagMAV_PROTOCOL_CAPABILITY_PARAM_FLOAT = ProtoField.bool("mavlink_proto.AUTOPILOT_VERSION_capabilities.MAV_PROTOCOL_CAPABILITY_PARAM_FLOAT", "MAV_PROTOCOL_CAPABILITY_PARAM_FLOAT", 33, nil, UInt64(2))
``` 
[image]
Although the documention suggests that this should be allowed:
Ref: https://www.wireshark.org/docs/wsdg_html_chunked/lua_module_Proto.html, section 11.3.7.14. ProtoField.bool

As mentioned, there points would need checking on a Wireshark developer forum, but I wanted to write them somewhere before I forget any details. 

Back to the Mavlink/pymavlink side of things.... as the high 32-bits of MAV_PROTOCOL_CAPABILITY and MAV_GENERATOR_STATUS_FLAG are not actually used, a 'fix' that have tested is to convert the UInt64 when calling the disector function, by adding ":tonumber()" to the value in these two cases:
```
dissect_flags_MAV_PROTOCOL_CAPABILITY(subtree, "AUTOPILOT_VERSION_capabilities", tvbrange, value:tonumber())
```  
This allows the dissections to work properly without errors - but all this will only work while the high 32-bits of them remain unused.

Its not perfect, but maybe its the best that can be done for now....

closed 2023-12-12 by peterbarker

hamishwillee, 2023-12-13:
Thanks @shancock884 - certainly a lot "better than nothing". I assume you tested that even though it won't work if the higher bits are used, it won't crash?

shancock884, 2023-12-16:
@hamishwillee - I did test with higher bits (notes in a previous comment).... the finding was that if ever higher bits do get used then Wireshark will throw an error when loading the LUA file. I did not try to address this in my change, and what I changed didn't make this any worse than it was before.
It is on my list of things to do to create a minimised version of the LUA script and raise it with Wireshark developers.

If you want, I could raise this in a new pymavlink issue to ensure the risk is captured for the future?
Or maybe this issue could be 'unclosed' as its probably fair to consider my fix as only a "for today" fix, as it did not address the entirety of the issue?

Thinking about it now I can think of 3 longer term 'fixes':
1/ raise it and wait for wireshark to be fixed, and then update mavgen with any change needed to work with it.
2/ workaround to ignore any bits higher than 32 and quietly not show them. (removes the LUA error, but data would be quietly missed from rendering, and probably noone will remember this and undo the workaround).
3/ split the field into 2x32-bit fields and render them as 2 separate blocks.

hamishwillee, 2023-12-20:
@shancock884 Thanks. 
If we were using https://mavlink.io/en/messages/common.html#MAV_PROTOCOL_CAPABILITY as it was intended it would already be full. "Fortunately" we realized it only had 64 bits and so have become overly frugal with using it (and we have no reliable solution for indicating support for features).
Upshot, I don't think we need to worry about this being used imminently, but it is possible.

I think we should do two things:
1. Do "1" - raise it in wireshark. We have enough timeframe to play with it.
2. Add a new issue "Wireshark Wlua dissection 64-bit flag fields crashes on higher bits" and link to this and the new issue you created above.

Does that make sense, and if so are you willing to do at least the first step of raising in wireshark?

PR #886 (2023-12-12): Mavgen WLua: Support 64-bit flag fields
Fixed Wireshark errors when decoding messages containing 64-bit bitmask/flag fields (fix #852).

Note that as per the notes I previously wrote in the issue, there is a wireshark limitation on the use of Uint64s meaning that this fix will only work while Mavlink bitfields do not make use of the high-32-bits.
This is currently OK for the only 2 messages using 64-bit bitfields (AUTOPILOT_VERSION and GENERATOR_STATUS)

I have also created a test case using a PCAP file containing an AUTOPILOT_VERSION message.