ArduPilot/pymavlink #864: Switching between versions of dialects

MauroPfister, 2023-10-10:
I was wondering if there is an easy way to switch between different versions of dialects in code. 

We are using a custom dialect to communicate with various devices over mavlink. Not all these devices necessarily use the same version of the dialect and we would like to be able to switch which dialect we're using depending on what device we are connected to.

Is there some way to achieve this?

I assume that I would have to generate the python modules for each version of the dialect according to [these instructions](https://mavlink.io/en/mavgen_python/#generate-a-custom-mavlink-dialect) and then make them available to mavlink by setting the `MAVLINK_DIALECT` and `MDEF` environment variables as mentioned [here](https://mavlink.io/en/mavgen_python/#dialect_file)?

peterbarker, 2023-12-14:
No easy way to do that.  You can include the new xml in `all.xml` and use the `all` dialect, however.

MauroPfister, 2023-12-14:
The problem here is that these dialects contain mostly the same messages but might be incompatible because some message definitions are different. I guess in this case I couldn't just include them in `all.xml` because the same message would be defined multiple times. Besides that, there would be no way of versionning the different dialects.

I recently discovered, that I can simply add the generated dialect `.py` files to the `dialects/v20/` folder of the pymavlink installation and then select that dialect with `mavutil.set_dialect("my_dialect.py")`. However, it doesn't seem possible to load these dialect files from another location (which would solve my problem). Is there any reason for this?