ArduPilot/pymavlink #873: CRC16_MCRF4XX checksum calculation broken?

maxfreu, 2023-11-10:
Hi! Could it be that the CRC implemetation [here](https://github.com/ArduPilot/pymavlink/blob/master/generator/C/include_v2.0/checksum.h) is broken? Disclaimer: I'm working on this for 20 minutes and have absolutely no idea about checksum computations, so this is a bold claim. But let's go through the code https://github.com/ArduPilot/pymavlink/blob/72ead371b5e7ad064577ecfd37cf8e083c71227c/generator/C/include_v2.0/checksum.h#L35-L43:

Here my remarks:

```C
static inline void crc_accumulate(uint8_t data, uint16_t *crcAccum)
{
        uint8_t tmp;  // note the initialization with 8 bits

        tmp = data ^ (uint8_t)(*crcAccum &0xff);  // take data byte and xor it with the low byte of crcAccum
        // shift the result from above 4 bits to the left and xor with itself;
        // this changes the 4 high bits and doesn't touch the lowest 4 (they are xored with 0s)
        tmp ^= (tmp<<4);

        // crcAccum >> 8 takes the high byte of the crc
        // tmp << 8 is just 0, because when we shift an 8 bit int 8 bits to either side, nothing is left
        // tmp << 3 and tmp >> 4 are ok
        *crcAccum = (*crcAccum>>8) ^ (tmp<<8) ^ (tmp <<3) ^ (tmp>>4);
}
```
So the parts I find strange are:
1) shifting tmp 8 bits to the left, leaving always zero and 
2) setting crcAccum to the xor of its high byte with an uint8, which means the result will always be 0x00XX. If someone could explain what's going on I'd be happy to learn!

Should tmp be uint16_t?

peterbarker, 2024-01-03:
Standard C arithmetic  integer-promotion rules apply here - basically your arithmetic operations promote to int (or uint) as the calculations are done.  The values will be truncated as they are actually shoved into storage as required.

Here's a test program:
```
#include <stdio.h>
#include <stdint.h>

int main(int argc, char *argv[])
{
    uint8_t tmp = 1;

    uint16_t crcAccum = 0;

    crcAccum = (crcAccum >>8) ^ (tmp<<8) ^ (tmp<<3) ^ (tmp>>4);

    uint16_t shifted = 0 ^ (tmp<<8);

    ::fprintf(stdout, "crcAccum=%u shifted=%u\n", crcAccum, shifted);
}
```

```
pbarker@fx:~$ g++ test-crc.cpp -o test-crc && ./test-crc 
crcAccum=264 shifted=256
pbarker@fx:~$ 
```

Gory details are over here: https://en.cppreference.com/w/c/language/conversion

I think we're good - not a bug?

maxfreu, 2024-01-03:
Ahh ok, I didn't know the values were promoted. Thank you for your explanations! :)

closed 2024-01-03 by maxfreu