ArduPilot/pymavlink #882: recv_match() error over UDP connection

jordancormack, 2023-12-02:
I've been using pymavlink to successfully send and receive mavlink messages using a serial connection, and I am trying to move to UDP.

When I try to receive a message using `msg = master.recv_match()` I get the following error:

"data, new_addr = self.port.recvfrom(UDP_MAX_PACKET_LEN)
OSError: [WinError 10022] An invalid argument was supplied"

However, if I send a message immediately before (e.g. `master.mav.heartbeat_send(...)` I don't get an error, and a burst of messages are successfully received, before a pause, and another burst of messages.

DevankGarg, 2024-07-16:
hi, did u figure out this ?

musluogluu, 2024-09-28:
I need help for this topic immediately. What do you mean by "I send a message immediately before (e.g. master.mav.heartbeat_send(...)" Can anyone help ???