ArduPilot/pymavlink #936: Utility to accept Mavlink messages and expose a Prometheus exporter interface

johann8384, 2024-04-12:
Would a prometheus exporter script using pymavlink be a welcome addition here? 

I've developed one for our in-house development but I thought maybe I could add the exporter and Grafana dashboards here. The idea is that QGC or other tools are setup to forward Mavlink messages to the exporter process which receives them and creates things like gauges for "satelites visible" and counters for distance traveled.

It allows us to have realtime dashboards in Grafana of our flights, to compare multiple aircraft or multiple flights of the same aircraft over top of one another. 

The dashboard and telemetry system is always running (Victoria Metrics + Grafana) so any time someone spins up an aircraft, SITL, bench flight controller, etc. they just automatically start logging metrics to the dashboards.

brunoolivieri, 2024-04-13:
I am not sure if I will use soon, but seems very cool and useful!  

Surveillance/Tethering guys might love it.

dermyrddin, 2025-08-06:
Hey @johann8384,
Did you have a chance to share your exporter and dashboard somewhere?