ArduPilot/pymavlink #948: License for C# Generated Code

tay10r, 2024-05-31:
Just want to clarify what the license is on the C# code.
Most of the generated code seems to be LGPL, but the project file seems to indicate that it is GPLv3.

peterbarker, 2024-06-01:
On Fri, 31 May 2024, Taylor wrote:

> Just want to clarify what the license is on the C# code.
> Most of the generated code seems to be LGPL, but the project file seems to
> indicate that it is GPLv3.

Did you read the README in root directory of the project?

Do we need to make it clearer?

tay10r, 2024-06-01:
Not sure if you're trying to be rude, but I'm just trying to be make sure the code is being used in accordance with the license in the repo.

The C# code contains a project file that describes the code (which isn't all generated) as being GPLv3. It might have just been a mistake, but that's really up to the copyright holder of the C# to decide.

tay10r, 2024-06-09:
Closing because I need to move forward and the protocol itself is documented well enough to come up with a different implementation, which I have already done. Cheers

closed 2024-06-09 by tay10r